prusa3d/Prusa-Firmware-Buddy #311: Unload Filament May Have Lower Probability of Breaking if Print Head is at Far Left

mlgold, 2020-03-22:
I had issues with the filament breaking on unload. I found that this did not happen if the print head is all the way to the left such that the Bowden tube is not as bent. Can others confirm this? Would it be possible to move the head to that position for load and unload? It would seem to provide the lowest friction in that position.

jweaver100, 2020-04-12:
I 2nd this.. My filament snapped in the tube yesterday and its no real surprise as it has such a tight radius.. If the head was over to the left, it would flatten out the tube a huge amount and reduce the likelihood of a break..

So I would like to see this implimented too.

mlgold, 2020-04-13:
As a workaround, I modified the custom gcode in the slicer after the print:

{if layer_z < max_print_height}G1 Z{z_offset+min(layer_z+140, max_print_height)}{endif} F720 ; Move print head up
G1 X178 Y180 F4200 ; park print head

It moves the print head far up and to the left, and the bed forward for easiest acess.

Eldenroot, 2023-01-11:
I think this one is also implemented @Prusa-Support

Prusa-Support, 2023-01-21:
Yes, I think it can be closed along with #1791.
Thanks for the heads-up.

Michele Moramarco
Prusa Research

closed 2023-01-21 by Prusa-Support