prusa3d/Prusa-Firmware-Buddy #313: min temp error

Lino77, 2020-03-24:
min temperature please lower from 15 to 5 degrees because the printer does not run below 15 degrees or is there a reason for it

x10102, 2020-03-25:
5 degrees seems kinda low to me, but I can't think of any problems besides the heater consuming more power.

DaSimme, 2020-03-25:
As far as I can see it could be even set to 0°C, since a disconnected thermistor shows as below zero temperature.
And that would be the fault to be detected by a min temp error.

Texy, 2020-05-13:
Absolutely. Prusa needs consistency between products - the mk3/s min temp was recently lowered from 15 to 10 degree's. I'd like the MINI to follow suit.

Lexorius, 2020-10-25:
ok, it is getting winter and my prusa mini wouldn't print in the garage. because it is only 14 °C. Please set the Temp to at least the same as the bigger brother. 
i hate to heat the printer up with the hotair gun befor printing...

PeterDHabermehl, 2020-11-21:
Guys, are you kidding me? My workshop will not be heated over night. When I started up two Minis this morining, I was not able to print because the thermistors showed 14 degC. Ridiciulous!

PeterDHabermehl, 2020-11-21:
> i hate to heat the printer up with the hotair gun befor printing...

Do it like Jesus - lay your hands on... until the bed reaches 16°C

Lexorius, 2020-11-21:
> > i hate to heat the printer up with the hotair gun befor printing...
> 
> Do it like Jesus - lay your hands on... until the bed reaches 16°C

You need like any other human with a nominal bodytemperatre over 35°C

Well. Bad news. The heat bad is really underpowered. It couldn't handle low temps around 10 °C bigger prints won't stick completle because of cold spots. 

I guess this is a really big designflaw

PeterDHabermehl, 2020-11-21:
I thought about bed insulation already. But that doesn't help much when the printer sits @15°C over night.

JohnnyDeer, 2021-02-16:
Min Temp has been lowered to 10 ° C, which is the lowest acceptable temperature considering machine limits. I'm closing this issue, thank you for understanding.

closed 2021-02-16 by JohnnyDeer