prusa3d/Prusa-Firmware-Buddy #322: Crashes when purging nozzle

kdpainter, 2020-03-26:
v4.0.4 crashes very repeatably at 95% when purging nozzle.  Have to reset to recover.

thejobbitt, 2020-03-28:
Also having this issue. It also happens in 4.0.5 RC1.

dahuafschmied, 2020-03-31:
I reportet this issue long time ago to Chat Support.

Vic-Y, 2020-04-11:
I have encountered this when purging after manually stopping a print (two times). When I ran purge after a reset (using display button), the purge completed successfully.

bobc, 2020-04-13:
Also happens in 4.0.5 final.

bildzeitung, 2020-05-07:
I’ve also observed this behaviour.

JohnnyDeer, 2020-06-22:
Hi, this issue is solved on 4.1.0-RC1. Please try it out :)

derenma, 2020-06-22:
Tested and LGTM!

(The flashing green cool down to 170/215C after leaving the purge menu confused me at first, but that is a really neat feature.)

JohnnyDeer, 2020-06-24:
Great, then i'm closing this issue as solved.

closed 2020-06-24 by JohnnyDeer