prusa3d/Prusa-Firmware-Buddy #334: M300 - play sound -> does not work

OndraCerv, 2020-03-31:
Hi, 

Code M300  not play sound.
Please of implementacion into new firmware. 

Thanks

hawwwran, 2020-07-01:
Yes, this!

matthewlloyd, 2021-06-28:
Anyone following this issue may be interested to know that I just released custom firmware that adds support for `M300` (play a sound), amongst other goodies:

http://github.com/matthewlloyd/Llama-Mini-Firmware#readme

fortinmike, 2021-07-08:
@matthewlloyd Did you consider submitting your changes as pull requests to the original repo? Many (including me) are not interested in installing custom firmware but would still love to see some of the things you implemented built in.

matthewlloyd, 2021-07-09:
> Did you consider submitting your changes as pull requests to the original repo? Many (including me) are not interested in installing custom firmware but would still love to see some of the things you implemented built in.

Given that Prusa has, to my knowledge, never accepted a pull request to this repo, apart from ones written by members of the Prusa team, I considered that approach to be impractical. Besides, they are IMHO violating Marlin's GPL by not releasing the source code for the bootloader, and I am not inclined to contribute to their open source efforts until they have addressed that issue.

If you dislike how difficult Prusa has made it to install custom firmware, I recommend getting in touch with Prusa directly to let your feelings about the appendix be known.

Meanwhile, you can always purchase a clone Mini or Buddy board, and run whatever firmware suits you on it.

fortinmike, 2021-07-09:
> Given that Prusa has, to my knowledge, never accepted a pull request to this repo, apart from ones written by members of the Prusa team

I was not aware of that. Are you sure that absolutely no pull request was ever accepted? Did they just ignore them or did they at least give a reasonable reason for refusing to merge? That's a bummer for sure. Still, maybe having pull requests sitting on GitHub might pressure them a bit and possibly help kickstart (possibly internal) development on specific features?

As for the appendix on the Buddy board, seems reasonable to me that they have some kind of way to know if, for example, you broke/bricked your machine because you installed defective, unsupported firmware on it. In the end they pay people to support users and IMHO it's not their job to fix issues you create yourself with modifications. Plus breaking the appendix only voids the warranty on the electronics instead of the whole machine. In essence, for sure no appendix would be better, but the fact that it exists does not bother me much, personally. On the other side of the coin, having a hardware mechanism that ensures only signed firmware is installed is a nice security feature to have.

To me, it's still incredible for a for-profit company to release all of their hardware and software as open source _and_ allow flashing custom firmware, with documentation on how to do it to boot, even if it's not perfect.

As for the bootloader, how do people make MINI clones without access to it?

Eldenroot, 2022-10-19:
Any progress?

Gordontgopher, 2023-01-01:
M300 is enabled in 4.3.4 but there are reports that it is buggy (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1075#issuecomment-1368260275)

Eldenroot, 2023-01-01:
I use 4.4.0 firmware.

M300 works fine at least for G-code start.

Not working for G-code end properly... 

I just use basic tones... so maybe it is bugged.

Would be really cool to fix this issue, even low cost Chinese printers can do that :)

Prusa-Support, 2023-01-05:
Pull Requests about M300 functionality were merged in January 2022 (https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1796, https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1800), supposedly only after the FW 4.3.4 final release (December 2021).

Same as https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/1075, thanks for reporting this.

Michele Moramarco
Prusa Research

bcolledge04, 2023-03-13:
Any update/progress on this ?

Prusa-Support, 2023-04-11:
No news. I'm sorry, [FW 4.4.1](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.1) was released shortly after the last few messages.
I'm afraid this issue didn't make it to the last firmware release. 

Michele Moramarco
Prusa Research

danopernis, 2024-07-09:
Let me close this as a duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3660 although this issue is certainly older. The fix is scheduled for next patch release. Sorry for all the confusion.

closed 2024-07-09 by danopernis