prusa3d/Prusa-Firmware-Buddy #348: FW update has no function

Lino77, 2020-04-06:
the FW update has no function under settings. when selecting, it always jumps back to the main screen

wffbos, 2020-04-10:
I can confirm this in the official 4.0.5 FW. Haven't verified if a valid FW-update file needs to be present in the USB-memory for the menu option to work. Did factory reset, but the problem persists.

x10102, 2020-04-11:
Can confirm it happens with the firmware on the USB stick as well.

pyro159, 2020-04-23:
I am having the same issue. It worked once but after I selected the option to flash firmware on restart, it didn't work again

AutomaticMFG, 2020-04-29:
Same issue on 2 Minis. 4.0.5 FW

nilsstreedain, 2020-05-10:
I'm having the same issue on 4.0.5 when going through settings or when forcing it to check the USB by holding the knob button down on startup.

x10102, 2020-06-24:
Fixed in latest release. Maybe close this issue?

JohnnyDeer, 2020-06-29:
Hi, as @x10102 wrote, in 4.1.0-RC1 FW Update function have description explained what it mean. I´m closing this issue as solved.

closed 2020-06-29 by JohnnyDeer