prusa3d/Prusa-Firmware-Buddy #360: Issue on Remote print - OctoPrint

Strankler, 2020-04-09:
Really nice feature, but with a bug.
The "remote printing in progress" is not only appearing while printing from OctoPrint, but always when OctoPrint is just connected.
So even if there is no remote job printing, you see this dialog, and have no possibility to access to the printer menu.
E.G for loading or unloading filament before or after a print, you need to disconnect OctoPrint and reboot the printer to access the menu for load/unload.

Would be great if that could be fixed.

x10102, 2020-04-11:
Maybe it's on purpose???

steve220, 2020-04-16:
This is true even when you disconnect from octoprint. Makes it difficult to change filament or start a print from the thumb drive.  One has to reboot the printer to access printer controls. Please change back to behavior under 4.04.

thsturgill, 2020-04-24:
It would be better for the system to revert to normal when the PI drops the connection. This was tested with both using the OctoPrint interface and shutting down the PI. Also would be nice to still have access to adjust live-z and to change filament.

Iqwertz, 2020-06-09:
Same Here! It would be great if Octoprint could send an Command at the Start of a print, to activate the blocking Octoprint screen and at the end of a print to have the normal menu during Idle.

Chleba, 2020-07-09:
Octoprint when it's connected to printer, will send gcode - "M876 P1". That code open our blocking screen.
Unfortunately when Octoprint disconnect from the printer, it will not send anything post-pone. You can always add your own custom gcode to send "M876 P0" or send it manualy to close this blocking screen.
You can always reopen it sending "M876 P1" manualy again.

bergdahl, 2020-07-29:
I got the following comment from foosel when I submitted this as a bug for Octoprint:

"It's very strange for the firmware to hook their "OctoPrint detection" into such a command to begin with, and frankly I'd consider it broken behaviour to be locked out of the regular printer menu just because a serial host is connected, at least not if that can be disabled."

For anyone looking for a workaround - Just set _Action Command Prompt support_ to _Never_ in _Settings > Plugins > Action Command Prompt_.

ysoldak, 2020-08-30:
Thank you @bergdahl !

This trick does prevent that limited and blocking OctoPrint screen from appearing (never liked it anyway).

For others looking for "Action Command Prompt" plugin, it is installed by default and is called "Printer Dialogs".
`Settings > Plugins > Printer Dialogs` and then "Enable support": `Never`

JohnnyDeer, 2020-10-21:
Access to the menu is solved in latest FW, I close this issue.

closed 2020-10-21 by JohnnyDeer

ysoldak, 2020-10-21:
I've installed latest released, v4.2.1 -- the issue is still there.
Perhaps Johnny meant it was fixed in trunk/master and not released yet...

reedy, 2021-02-04:
> I've installed latest released, v4.2.1 -- the issue is still there.
> Perhaps Johnny meant it was fixed in trunk/master and not released yet...

Yeah, it'd be useful to have a reference commit/issue... It might be in 4.3 maybe? "latest firmware" is not helpful to anyone; 4.2.1 was the latest at the time, and apparently wasn't fixed. @JohnnyDeer want to clarify?

I see 4.3.0 is now stable as of today... So I guess we can more easily double check this now (my printer is current printing, so will update later on today).

I've ended up here after being annoyed by plugging in (connecting; physically or in software) an octoprint pi, and the GUI being blocked.

It seemed odd that it did this if not printing, and then you couldn't kick off a print using the prusa mini screen/menu, but if you are printing it just went into "monitoring mode" (which is what I wanted; I have no desire to print from octoprint atm)

reedy, 2021-02-04:
Yeah, so this doesn't seem to be fixed on 4.3.0 either (at least on the Prusa side; I've not changed the OctoPrint plugin setting as suggested above).

While there's not a print in progress, plug in the Pi, connect OctoPrint, it's "stuck" on the "Serial Printing" window (bar hitting tune), unless I hit disconnect...

Similarly, if I unplug the Pi... It stays on the screen, so it doesn't detect the serial disconnected (physically), nor if I just click "Disconnect" on the OctoPrint UI