prusa3d/Prusa-Firmware-Buddy #361: Colorchange dialog asks for filament insertion before preheating nozzle

Eisfoehniks, 2020-04-11:
I recognized that when adding color changes in prusaslicer, the printer will stop as expected and cool down the nozzle. However it will directly ask to insert new filament and press continue.
If continue is pressed, the interface freezes in the "inserting" stage until the nozzle reached temperature and then automatically start pulling the filament. This way the filament needs to be constantly pushed into the extruder until teamperature is reached. 
Asking for user confirmation before actually pulling filament would be more convenient.

CyberHugger, 2020-04-17:
I would prefer that the printer not cool down immediately on a color change. There should still be a timer but give us a chance to swap out the filament without waiting for the printer to warm up again. If we are making the filament swap when the printer is cool, why can’t the insert phase occur while the printer is heating? I would also like to see the printer ask if the extruder grabbed the filament. Then repeat the insert until we say yes.

scubagregory, 2020-06-09:
I would also prefer this work like the MK3S and not cool down so quickly or allow a menu option to set the cool down time.

JohnnyDeer, 2020-06-29:
Hi, filament changing function have changed behavior in FW 4.1.0-RC1. If temperature drop down (after 30 second) filament is loaded to gears immediately, before reaching temperatures, and load will be done automatically after their reaching. Thorugh this change is "wait for temperatures" issue avoided. I´m closing this issue as solved.

closed 2020-06-29 by JohnnyDeer