prusa3d/Prusa-Firmware-Buddy #362: Remote printing w/ Octoprint - filament change not working

Isaac12x, 2020-04-12:
Further to issue #360 during remote printing when the printer reaches the M600 command in GCODE
extruder pulls out the filament and waits for user input (a.k.a. continue when the new filament is loaded).

So, while doing that, when the new filament is loaded and I press continue, the printer comes back to the remote printing screen without loading the filament. Funny thing is it doesn't come back to print either.

d-sko, 2020-04-15:
Same here, also with FW 4.0.5 and PrusaSlicer 2.2.0. Seemingly there is no way to resume the print after the filament is unloaded.

Isaac12x, 2020-04-18:
I've dug a bit more in the issue trying to execute the G-code sequence manually. However, the receiver seems to be stuck in some sort of loop.

I guess it's expecting user input (press continue button when the new filament is loaded) but not registering the button press.

Message is:
"Recv: echo:busy: paused for user"

derpicknicker1, 2020-05-12:
Same problem here...

davetyra, 2020-05-13:
This may be related, M601 M602 as inserted in the preview screen of PrusaSlicer 2.2.0 has similar behavior, printer pauses, but will not resume when button is pushed, this is printing from USB, not remote. FW 4.0.5 and PrusaSlicer 2.2.0

Iqwertz, 2020-06-09:
I have the same Problem!

JohnnyDeer, 2020-06-24:
Hi, in new FW version 4.1.0-RC1 is M600 repaired for printing via OctoPrint. Please try it out. I´m closing this issue as solved.

dwineman, 2020-06-24:
Just tried it — the problem has not been fixed. Filament unloads, printer returns to the Octoprint screen. Never prompts to load new filament. "Change Filament" in the Tune menu does nothing. It is impossible to resume the print.

ifekali, 2020-06-24:
Same here, not fixed yet.

Crowlord, 2020-06-25:
Confirm not fixed. Multi coloured prints have to be done from USB stick.

deltwalrus, 2020-06-26:
Confirmed here as well, not fixed.

JohnnyDeer, 2020-06-30:
I can confirm too. Sorry to all, lesson for me... test everything by myself.

JohnnyDeer, 2020-07-08:
Finally this issue should be fix in 4.1.0 final. Please try it and confirm that it is fixed :)

olsw, 2020-07-08:
Can confirm this is definitely working now. Thanks!

danielo515, 2021-02-07:
> Finally this issue should be fix in 4.1.0 final. Please try it and confirm that it is fixed :)

4.1.0 of which software? Prusa-slicer? Octorpint? Something else?

danielo515, 2021-02-07:
Whatever it was, it's working for me now, haha

s-taylor, 2021-11-17:
So I’m on the latest firmware 4.3.3 and this did not work for me. I have since done a factory reset and now it works for me, hope this helps someone else.

twobit0, 2022-11-07:
Still Not addressed on 4.3.4

Crowlord, 2022-11-07:
Why was this ever closed. Its never been fixed?

danielo515, 2024-06-14:
Stale doesn't mean fixed

danopernis, 2024-06-14:
Stale indeed doesn't mean fixed, but it does mean that the issue did not receive any updates in a long time.

@danielo515 can you test if the issue is still present in the newest version of the firmware? There have been several improvements and bugfixes since 4.0.5

closed 2024-08-22 by github-actions[bot]