prusa3d/Prusa-Firmware-Buddy #363: Change Request: Change behavour after print

jweaver100, 2020-04-12:
After you print, you get the option to TUNE/REPRINT/HOME, which is nice IF you intend to print again

But often I don't want to print again and accidentally hit REPRINT instead of HOME... This then causes the printer to start warming and if you hit stop, it does nothing.. Certainly not for a few mins.. So when this happens, I generally have to press 'reset' so I can do what I want.

Could we get a tweek to this functionality.. Some options

1. HOME is pre-selected rather than REPRINT - If you want to reprint, you simply have to move and select.. But if you don't, there is no harm done as you just go to HOME

2. Add a "confirm" to reprint.. So if you select it, it says "this will reprint again, are you sure"

3. Change the "Stop" behavior so that it actually stops quicker, if the print hasn't already started.

Personally, I like option 1 myself... For people who use "reprint" a lot, this isn't a massive impact (one extra step) and if you make a mistake and select HOME there is very little impact.

x10102, 2020-04-12:
Great idea! I'd personally like all 3 options as the third one will be useful not only when reprinting.

jweaver100, 2020-04-12:
Sorry.. I didn't realise it had already been posted.. I have been thinking this since the day I got my Mini and assumed that because it was a quick fix and not already implimented, no-one had reported it.

I can see that against issue #26 there is a commit, so hopefully this will be fixed soon.

peros550, 2020-05-13:
All these are good ideas. I would also like to add something because it has to do with after print behavior. 

When print finishes, Prusa mini moves the bed all the way towards the user, like serving a "glass of wine" to the user. 

Well, nothing wrong with that, very clever indeed, but sometimes the filament continues to oozle and it fills the top right corner of the bed , very close to a sensor which sits under a black plastic cover. 

Perhaps that movement could be tweaked to allow some space for the filament.

jweaver100, 2020-06-23:
I tried 4.1_RC1 yesturday hoping that this might have been fixed.. After all, the fix should be REALLY easy.. I would just be happy if the default option was "HOME" rather than "REPRINT".

Issue #26 was mentioned above as a duplicate.. And looking at that issue its says that there is a commit, so I assume this ment that it had been fixed... 

Is there an intention to change this behavior? I hope so as I still kick of a reprint by accident regularly......

Claghorn, 2023-11-03:
Making the default "Home" would be very nice, impossible to accidentally hit "reprint" then.

closed 2024-07-04 by github-actions[bot]