prusa3d/Prusa-Firmware-Buddy #364: New EEPPROM implementation unhandy for extruder upgrade

DrSchumax, 2020-04-13:
As the extruder of the Mini is not very great and bondtech now offers a really good and simple drop-in upgrade to a dual drive extruder a lot of people are going to do this upgrade once the word spreads. So far this worked without problems.

Unfortunately with firmware 4.0.5 it is no more possible to adjust the E-steps (M92) in the firmware as it will always be reset to the default. Also an adjustment for filament unloading length is no more possible to release the filament (M603).
Setting the ESteps in the gcode is possible but very unhandy as it requires to change the original prusa profiles every time, for M603 this is not possible or will require a very weird workaround with running a gcode every time upon startup or so.

I understand that it is unhandy to have 2 EEPROM, but there got to be a smarter solution than this that allows users to make adjustment to basic parameters without requiring a custom firmware. If it is possible for the PID and Z-axis adjustment it should also be possible for those 2 mentioned options.

tangero, 2020-04-24:
Hi Dr!

As of version 4.0.5, we turned off EEPROM support in Marlin because values were written inconsistently. All parameters can be changed using GCODE, but will have a default value after reboot. The EEPROM now only remembers the values that are part of the calibration (Live Adjust, PID) and still the Ethernet settings. These parameters are written separately.

We do not plan to turn on EEPROM support, but we will probably add other parameters. If the user has a modified extruder, it is advisable to change the parameter in GCODE or recompile the firmware.

closed 2020-04-24 by tangero

DrSchumax, 2020-04-25:
@tangero sure because you guys have made it so easy to run unsigned firmware. Can you please elaborate on the issue with data inconsistency because from the explanations I have seen I do not understand why simple writing and reading data from non-volatile storage would be so immposible that it justifies to castrate Marlin in this way?

khoi, 2021-07-30:
+1 for persisting extruder estep to EEPROM, this is a much needed addition and shouldn’t require a custom firmware.