prusa3d/Prusa-Firmware-Buddy #365: Printer crashes when preheating in low temperature

deepthought-64, 2020-04-13:
When trying to preheat the printer at a temp of 13°C (assume it does it below 15°C) the printer firmware crashes and gets reset by the watchdog. 

Expected behaviour would be that it should either:
- show a warning that it is too cold to pring (which i personnally dislike, because why should i not be able to print at 13°C?)
- show a confirmation dialog telling me that it is cold but lets me continue, or
- just let me print (I'd perfer that)

See video here: https://photos.app.goo.gl/DXRSgSNTFWSDb93r5

I'm using firmware version 4.0.4

x10102, 2020-06-24:
Yes please. I want this.

Hl2run, 2020-06-24:
Same here please.

Vic-Y, 2020-07-23:
I think I have a related problem with firmware 4.1.0. My printer is in a cool room in my house, under an air vent. With the air conditioning on, the printer temperature is below 15C when idle. When I select "unload filament", printer displays an error message and forces a reset.

[image]

[image]

JustAnother1, 2020-07-27:
The problem is probably connected to a safety issue. If the temperature sensor is disconnected or short circuited the temperature read will probably be around what you measure.  To avoid burning down your house the firmware tries to detect these error conditions (That is what the MINTEMP is about)

If you really need to print at those low temperatures then you can change the MinTemp setting in the firmware.

JohnnyDeer, 2020-12-14:
Hi, in version 4.3.0-RC1 mintemp is reduced to 10 ° C, which will solve this issue. @deepthought-64 can you confirm it please?

deepthought-64, 2020-12-14:
HI @JohnnyDeer, thanks for the notification. Will update and test it!

deepthought-64, 2020-12-14:
I can confirm, that pre-heating works now below 15°C but however if the printer is below 10°C it still crashes when trying to preheat. 
I expect that instead of going into a watchdog-reset it should just shows a dialog telling the user that the temp is too low. Instead it resets and shows a mintemp error and instructs me to check the printhead thermistor for damage in the wiring.

[image]

JohnnyDeer, 2021-02-10:
Mintemp lowered to 10°C is minimum we can do given the machine limits, values are unified with the MK3 series. I´m closing this issue, thank you for understanding.

closed 2021-02-10 by JohnnyDeer