prusa3d/Prusa-Firmware-Buddy #366: Total print statistics

josecoelho96, 2020-04-14:
Similar feature to others like the MK3, show a total summary of some statistics:
- Total Print Time
- Total Filament
- Total Number Prints

Just for enhancement and curiosity.

x10102, 2020-04-14:
Totally would like to see that!

iandouglas, 2020-04-24:
I'd love to see these stats even if printing with tools like OctoPrint. I've printed miles and miles of filament through OctoPrint on mhy MK3/MK3S as well, but none of it gets reported via firmware unless printing via USB. Would be nice to see this on the Mini as well.

4nthonylin, 2020-08-24:
Any update on this? +1 For this. Another great feature with stats would be the ability to inform users of maintenance.

For example:
1) Every 100 hours, inform users to clean/lubricate rods, and then confirm when done. Printer can internally track next cycle
2) Check belt tensions
3) Replace any consumables (PTFE Tubes) etc

Basically something like a car maintenance cycle.

hemdrup, 2020-12-19:
Yes, this would be cool.

DanidCH, 2021-07-03:
Even if the calculation isn't 100% perfect this would be of great help to know when is a good time to do maintenance!

JohnnyDeer, 2021-09-08:
Hi, this feature is implemented in new 4.3.3-RC. You can try it out https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.3.3-RC.

Eldenroot, 2022-11-26:
@Prusa-Support - this is done :)

Prusa-Support, 2022-12-10:
Closing. Thanks 🙂 

Michele Moramarco
Prusa Research

closed 2022-12-10 by Prusa-Support