prusa3d/Prusa-Firmware-Buddy #367: Feature Request: Support for PINDA v2 and Z-axis default esteps 200

bbeller123, 2020-04-14:
It would be nice to have a bed leveling probe like the MK3S on the Prusa Mini. This would improve bed leveling consistency.

If the z axis motor has to be replaced and the pitch is not correct, the esteps for Z-axis must be updated. The MK3S and Mini do not have the same pitch.

ogland, 2020-08-09:
I don't understand this. The mk3 and Mini uses the same leadscrew/pitch and the Z has zsteps, not esteps. Only the E axis has Esteps.

The prusa mini already has a bed leveling probe, called the MINDA. Only difference between that and Pinda2 is the embedded thermistor. Reason for not having pinda2 seems to be that MINDA is lighter and cheaper, and it's a printer that's supposed to be cheaper so of course it isn't gonna have the same features or quality.

bmangodt, 2020-08-14:
@ogland The Buddy board is built to accommodate a Pinda v2 sensor. It would be nice to have the firmware act upon this. If users don't want to spend the extra money on a Pinda, they don't have to. But if they want to and it's clearly ready to support it, then why not? I'm not willing to void warranty and built a custom firmware just for this, but it would be nice to have it officially supported. I'm all for it!

ogland, 2020-08-14:
@bmangodt I agree

Prusa-Support, 2022-05-22:
Hello! Sorry for intervening on this thread so late. In the end, while this very feature for PINDA V2 support on MINI wasn't released, we currently have the SuperPINDA probe, which is the same between MINI+ and MK3S+: https://www.prusa3d.com/product/superpinda/. Closing the issue for now based on this aspect. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-22 by Prusa-Support