prusa3d/Prusa-Firmware-Buddy #374: [BUG] Remote printing tuning

Isaac12x, 2020-04-18:
When using the remote printing option introduced in RC1 FW v4.0.5, where you can see the Octoprint logo on the screen this is what happens.

Remote printing screen --> tune --> change filament --> Nothing happens.

jurcja1, 2020-04-20:
Same issue but during printing from USB. During print "change filament" - nothing happens.

tangero, 2020-04-24:
Duplicated to #362

tangero, 2020-04-24:
Hi Isaac,
Try please Factory reset. It looks like you have heating off.

Isaac12x, 2020-04-26:
Hi @tangero, I already did that. Let me be more explicit.

While it is remote printing and the machine hits an M600 instruction, the filament is unloaded but you can't load a new one.  So in theory heating is on as the filament which was printing with gets unloaded.

Any other ideas?

derpicknicker1, 2020-05-12:
Same problem here

aaaaron, 2020-06-03:
Same problem, M600 issued through Octoprint never succeeds.

JohnnyDeer, 2020-07-09:
Hi, this bug is fixed in 4.1.0, please try it out. I´m closing this issue as solved.

closed 2020-07-09 by JohnnyDeer