prusa3d/Prusa-Firmware-Buddy #377: I _think_ the z axis live adjust stopped working in 4.0.5?

glitchykitten, 2020-04-20:
I was running 4.0.5 RC1 without any problems.  Since upgrading to the final release of 4.0.5 I think I've noticed that the z axis adjustment doesn't "live adjust" anymore.  This happens both during first layer calibration and using the live adjust functionality during a print.  If I stop a print and restart it, the new z offset value does correctly take effect.

I've also started to have issues with the first layer calibration routine hanging the machine after the mesh bed probing completes.

Is anyone else seeing this?

Isaac12x, 2020-04-21:
Yes, same here. I also find some interesting behaviour when I run the first layer calibration. It was working well at 4.0.4.
After updating to 4.0.5 - when the nozzle carriage is getting towards the leftmost side of the bed it lifts itself up automatically.

lensdigital, 2020-04-23:
It's just the opposite for me.  Live Adjust didn't work until I upgraded to 4.0.5 and also did factory reset. Not sure which part did it. Try factory reset...

kurkowskik, 2020-04-25:
I also upgraded to 4.0.5, did factory reset and everything works fine.

JohnnyDeer, 2020-05-12:
Hi, @glitchykitten did you tried factory reset? Is mesh bed leveling work correctly? 
@Isaac12x your issue looks like Minda problem, please see forum, if behavior of your printer is similar ( https://forum.prusaprinters.org/forum/hardware-firmware-and-software-help/z-axis-homing-issue/ )

x10102, 2020-05-22:
I'm also having calibration problems on 4.0.5

JohnnyDeer, 2020-05-25:
@x10102 could you please describe it? Did you tried factory reset?

x10102, 2020-06-02:
> @x10102 could you please describe it? Did you tried factory reset?

So the problem is that even though I calibrate it nearly perfectly one time, it's either too close or too far from the bed when I want to use it again in a few days. Also it kinda seems to me that the print head is higher when it's on the far left side.

I have tried factory reset two times and PINDA is working and at the correct height.

JohnnyDeer, 2020-06-09:
@x10102 your description seems related to #309  , do you agree?

x10102, 2020-06-14:
@JohnnyDeer Yes it is, but it seems the problem went away, not sure what happened.

x10102, 2020-06-18:
@JohnnyDeer Okay it happened again today. It's really inconsistent

x10102, 2020-06-18:
@JohnnyDeer Someone in #309 said something about waiting a few minutes between warming up and printing fixed the problem for him. Now that I think of it, it might be a solution for me as well because I mostly warm up the printer before printing and it sits for a few minutes, but I didn't today and that's when it happened. Going to test this.

x10102, 2020-06-18:
@JohnnyDeer Confirmed it's related to preheating. If I leave it sitting at 215/60 for a few minutes before printing it prints perfectly with around -0.58 offset. If I don't it's all over the place and needs readjusting every time.

hawwwran, 2020-06-30:
> @JohnnyDeer Confirmed it's related to preheating. If I leave it sitting at 215/60 for a few minutes before printing it prints perfectly with around -0.58 offset. If I don't it's all over the place and needs readjusting every time.

Confirmed here

JohnnyDeer, 2020-06-30:
Thank you for confirmation @x10102 @hawwwran . I think #636 is also related to this problematics...

x10102, 2020-07-01:
@JohnnyDeer I think it got worse in 4.1.0 RC1. Not even the preheating seems to fix it now.

JohnnyDeer, 2021-02-16:
This problematic has been solved by releasing SuperPINDA, more informations is in #636. I´m closing this issue.

closed 2021-02-16 by JohnnyDeer