prusa3d/Prusa-Firmware-Buddy #388: G-code for reading filament settings

urholaukkarinen, 2020-04-25:
There does not seem to be a way to read the current filament settings from printer. 
Perhaps it could be done with M503 g-code, but at least for Prusa Mini it reports `echo:Filament settings: Disabled`.

When a filament is changed to printer, the filament type must be specified. Filament type is also specified in PrusaSlicer. It is possible to accidentally print with the wrong filament settings, so I have been thinking about writing a OctoPrint plugin that checks whether the current filament type matches the gcode's filament type (temp settings).

JohnnyDeer, 2020-05-06:
Similar theme as #403

JohnnyDeer, 2020-10-21:
I close this FR for same reason as #403 thank you for understanding.

closed 2020-10-21 by JohnnyDeer

HarkonenBade, 2024-04-16:
Is there any hope that this might be reconsidered? As I would like to develop an octoprint plugin to validate the filament type, and reading it from the printer would be the ideal way to check what is currently loaded.