prusa3d/Prusa-Firmware-Buddy #408: MINTEMP - Reset Printer

AutomaticMFG, 2020-04-29:
Received error printing PLA (12 hours into a 15 hour print). 

Error details:
"The requested Bed temperature was not reached.
Nozzle temp: 210/210
Bed temp 53/60"

This threshold is too narrow and should pause the print to allow temp to recover.

//Reduced feedrate on this print to 80%
//New Mini, have checked cables to heatbed and TC and do not see any variations in reported temps.

x10102, 2020-05-11:
What's your room temperature? The ERR_MINTEMP error used to happen to me when my room was too cold.

JohnnyDeer, 2020-05-12:
Hi, @x10102 that's good point, room temperature have big impact to  mintemp, also wind from open window can trigger it. @AutomaticMFG can you confirm any of this situations, or it´s happen in common conditions?

pjenkin, 2020-06-11:
OT (apologies), but as a noob i saw this message in the calibration wizard when first setting up a Prusa Mk 3 Mini - in my case, tightening the cable clamp on the buddy board/electronics box (step 28 in the handbook) had pulled out the connection for the heatbed heater (step 24), hence the bed not heating. Reconnected, all ok.

AutomaticMFG, 2020-06-11:
On our system the issue proved to be a faulty cable to the thermistor. The issue got worse as time went and it was easier to determine root cause. New sensor resolved the issue. 

Thanks

JohnnyDeer, 2020-06-15:
@AutomaticMFG good to hear, that you found problem and is already solved. I'll close this issue as solved.

closed 2020-06-15 by JohnnyDeer

fabthefabulousfab, 2020-10-08:
Hi @AutomaticMFG 

I just got the exact same issue, symptom and context, could you indicate me more precisely the cable that you changed : is it the red/black/green/red cable that is going from below the mini ?

Thks!

JohnnyDeer, 2020-10-12:
Hi @fabthefabulousfab please contact our Support with this issue. Chat in right down corner https://shop.prusa3d.com/en/ . Thank you.