prusa3d/Prusa-Firmware-Buddy #410: Wrong Z axis value after self test

urholaukkarinen, 2020-04-29:
I did a self test on my printer and immediately after that went to settings to manually move the axis. Z was somewhere around 240 and when I changed the Y value, the Z axis made a creaking sound. After I started changed the Z value it immediately changed to 170.

JohnnyDeer, 2020-05-05:
Hi @urholaukkarinen , thank you for reporting. With which FW version this happen?  Did it happen to you just once, or it's repetitive problem?

urholaukkarinen, 2020-05-05:
This occurred with FW version 4.0.5. I have not used the self test since, so this has only occurred once.

JohnnyDeer, 2020-05-12:
Issue reproduced and confirmed.

dkrieger, 2020-10-03:
this issue affects me as well, on both 4.1.0 and 4.2.1. In my case, the erroneous z value is 203 after self-test. If I don't auto-home after self-test, the next MBL finishes by digging the nozzle into the corner of the sheet, damaging it in the first days of usage and until I figured out how to avoid it.

not sure if related or not, but when I power on, MBL (manual or otherwise) barely moves the head left when it goes through its paces. again, auto-home first usually prevents this.

closed 2024-07-04 by github-actions[bot]