prusa3d/Prusa-Firmware-Buddy #412: WiFi module support

feutl, 2020-04-30:
I would be interested in which WIFI Module can be used with the new Buddy Board.
I do not have any ethernet cabeling next to my Mini, but would like to get access to the WebUI.

Linking another ticket asking the same question in a different repository
https://github.com/prusa3d/Original-Prusa-MINI/issues/8

martin8425, 2020-04-30:
Have a look for "ESP8266 ESP-01 Wifi module" on eBay. You can get them for just a few bucks.

But I don't think the software is implemented yet, is it!? I can't get mine to work.

martin8425, 2020-04-30:
It might be in the works. Here's a screenshot from the 4.0.5 release notes with the wifi symbol up top!! 😮 

[image]

Hustenbonbon, 2020-05-25:
Any update on this?

MotStr, 2020-07-29:
Any update? I will buy the printer as soon as WiFi is supported 👍

spheppner, 2020-07-29:
Any update yet? This has now been dragging since april

probonopd, 2020-07-30:
If you are a tinkerer, you might want to check out https://github.com/probonopd/WirelessPrinting and see if you can get it going on the Prusa Mini (also see https://github.com/probonopd/WirelessPrinting/issues/157).

trondkla, 2020-10-15:
Any update yet?

MotStr, 2020-10-15:
Is there at least any schedule for this? I feel like it was promised since the beginning, but now it's getting postponed all the time.... I'm happy I haven't bought the printer yet, becuase I would be feeling cheated now :D

probonopd, 2020-10-19:
If I had access to such a printer, it'd probably be implemented by now ;-)

spheppner, 2020-10-20:
It's getting kind of ridiculous now, since this issue has been postponed since it was opened. Please respond

Hustenbonbon, 2020-10-22:
It is a bit ridiculous as Prusa made it look like this is a feature coming soon after release, and now they are waiting until someone from the community builds it. I mean it is cool that this is possible at all, but Prusa definitely overpromised here.
I am just continuing to use octoprint for now.

jhenley01, 2020-11-12:
I agree with the previous posts. They did make it clear that wifi would not be available upon release but here we are over a year later and none of the promised features like wifi, power loss recovery or far software have been added. In fact we've actually lost features as saving settings to EEPROM was removed in a recent update.

ZeekWorld, 2020-11-28:
SuperPINDA but still not Wi-Fi support... Almost a year now since Mini release.

drakor007, 2020-12-10:
PINDAt  (pindat .... verb)  means in the Czech language "have boring/annoying speech". Super PINDAt would be the next level.
So, I stop believing that this functionality will ever work....

padigree, 2020-12-20:
Is there any schedule for this? Still waiting now since nearly a full year for this feature - would be glad to use native wofo support instead of octoprint :/

merdok, 2020-12-20:
Also waiting for this...

riodoro1, 2020-12-24:
Everyone is waiting guys, please, make it happen!

padigree, 2021-01-22:
Yeah would be glad.. There is much going on in the firmware at the moment, but haven't heard the word wifi for long time..

Eisfoehniks, 2021-02-01:
Hopefully it's coming soon. This pull request looks promising: https://github.com/prusa3d/Prusa-Firmware-Buddy/pull/1246

Noxmiles, 2021-03-11:
Hello @josefprusa - say something. Need Wifi on the Mini. Don't need the 3mf file format.

asdwsda, 2021-03-12:
Stay calm guys! WiFI support is definitely under development.
[image]

Hustenbonbon, 2021-03-12:
Hey @asdwsda , sounds like great news, from which repo are these branches? They are not from this repo.

Still, after almost a year, any kind of response from Prusa would be nice.

asdwsda, 2021-03-12:
Insights / Network and check the Prusa developer's forks 😉 

> Hey @asdwsda , sounds like great news, from which repo are these branches? They are not from this repo.
> 
> Still, after almost a year, any kind of response from Prusa would be nice.

asdwsda, 2021-03-12:
I also seen traces of buddy firmware for MK4 (and crash detection for the MINI), but it is moved to private repo.

jamesjnadeau, 2021-11-16:
to anyone listening, I'm an experienced developer willing to test this out in the field, I have a mini, I'll buy the required esp board if someone points me at a link.

Niflheim123, 2021-11-16:
search for ESP8266 ESP-01

trondkla, 2021-11-16:
For example
https://www.banggood.com/ESP8266-ESP-01-Remote-Serial-Port-WIFI-Transceiver-Wireless-Module-p-947259.html

0xygen8, 2021-11-16:
rip-off: https://www.aliexpress.com/item/32582736130.html?spm=a2g0o.productlist.0.0.42dc5cd5ERpMWb&algo_pvid=c13c4534-05f6-4757-9723-3723ec4a2567&algo_exp_id=c13c4534-05f6-4757-9723-3723ec4a2567-6&pdp_ext_f=%7B%22sku_id%22%3A%2210000015479862662%22%7D

jamesjnadeau, 2021-11-16:
Thanks folks, board ordered :)

looks like this is the current tree with the latest code for the wifi board:
https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/ESP8266

someone correct me if I'm wrong.

Emanuelgtxf, 2021-11-18:
E sera que isto funciona prusa mini muita parra pouca uva

riodoro1, 2021-11-19:
Now that the Prusa XL has been announced I wonder what the chances are that Prusa will improve and maintain this branch of code. They didn't even advertise the XL as having WiFi support.

Now that #1698 has been merged will there be any communication from Prusa about the state of implementing WiFi module support on the buddy firmware or are we bound to guessing from the contents of the commits and speculate on "private repos"?
Is there even a roadmap for this product?

jhenley01, 2021-11-19:
> Now that the Prusa XL has been announced I wonder what the chances are that Prusa will improve and maintain this branch of code. They didn't even advertise the XL as having WiFi support.
>
> Now that [...]

I’ve asked Prusa support several times over the past few years when WiFi support was coming and the answer was always “very soon” so I don’t think Prusa even knows at this point. 

Also kind of a slap in the face that the XL has power loss recovery and the Mini still doesn’t have it.

riodoro1, 2021-11-20:
> Also kind of a slap in the face that the XL has power loss recovery and the Mini still doesn’t have it.

It might be because the XL’s power supply has a power loss warning pin and can tell the processor to use whatever’s left in the caps to save the state.
Without such signal the cpu has to basically log it’s position someplace non-volatile all the time.

I think however I remember Prusa promising this feature for the mini.

mflamm, 2021-11-20:
> Thanks folks, board ordered :)
> 
> looks like this is the current tree with the latest code for the wifi board: https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/ESP8266
> 
> someone correct me if I'm wrong.

Awesome James! I’m looking forward to any news from your side :)

riodoro1, 2021-11-26:
> > looks like this is the current tree with the latest code for the wifi board: https://github.com/prusa3d/Prusa-Firmware-Buddy/tree/ESP8266

I think this branch is merged, so either WiFi is working on master or (as I suspect) this was some internal fix and we still have to "wait a bit" for a full support.

jamesjnadeau, 2021-11-30:
I stopped working on this because at the end of the day I can't upload gcode, which effectively makes it not worth it to me. My time is better spent setting up another octoprint server.

I was able to get master to compile, it didn't produce a bbf file, but that's where I stopped.

trondkla, 2021-11-30:
Thanks for trying James

RichiH, 2022-08-07:
This issue can be closed now, I presume? I installed the module yesterday and it works.

https://help.prusa3d.com/guide/how-to-setup-wi-fi-mini-mini_316781

Prusa-Support, 2022-08-13:
Thank you all for your patience and trust.

Recently it was released [FW _v4.4.0-beta2_](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0-beta2), the first publically available firmware compatible with ESP WiFi module. As you can see, the release notes are quite extensive, and somehow explain the difficulties and delays of the WiFi implementation.

In the section _Hardware prerequisites_ you can find recommendations about the ESP module, and a link to the spreadsheet of tested ESP modules.

In the section _Step by step guidance_ you will also find links to the instructions and the Forum guidepost.

I hope you don't mind closing this issue now. However, in case of further related questions, don't hesitate to ask. 

Michele Moramarco
Prusa Research

closed 2022-08-13 by Prusa-Support