prusa3d/Prusa-Firmware-Buddy #414: Crash or Stall Detection on Z Axis

aurelw, 2020-05-02:
So I ran into the following issue, also clear user-error.

I discovered the printer paused in the middle of a job with the hotend already cooled down. The firmware crashed and showed a watchdog error message. Confirming it, natuarally restarts the printer. However, the nozzle is stuck to the part. I feel a bit lost and think about continueing the print like after a crash or filament sensor trip but realize that this won't be possible. So I want to get the part off the platform. It's late in the evening, never moved the axes on the mini manually before and somehow I am out of my mind think, auto-homeing sounds like a good idea. So, what happened,....

The printhead lifts from the platform with the part still stuck to the nozzle, it homes on the x-axis, on the y-axis, and finally slams down with full force on the z axis at the corner of the build platform, bending it almost one centimeter down. Only pressing the emergency-reset button stops the printer.

My suggestion, since stall/crash detection is done on x and y axis, doing something similiar on the z axis may add some safety and may prevent damage to the printer or someones hands.

FIrmware Version: 4.0.4

JohnnyDeer, 2020-05-05:
Hi @aurelw , thank you for your suggestion. Crash detection is not implemented in Buddy FW at this time. Best way to solve this situation is heat up hotned and move Z up in Settings/Move Axis.

mlgold, 2020-05-07:
Another scenario is if you accidentally hit select reprint, then immediately hit stop, it still moves the head down to crash into the last item printed before actually stopping. It ignores any commands, so you cannot heat the axis up or move the Z axis up. Only hitting the reset button will stop it.

JohnnyDeer, 2020-05-12:
Movings, when is selected stop print, immediately after start is known issue. User should be unable hit  reprint while watchdog reset. Ideal scenario is reset, preheat, wait for temp, them settings/move axis/lift Z up and manuali remove stucked print.

sergiusz-roszczyk, 2020-05-16:
Hi. Today I also got "smash into build plate" issue. After around 1 - 1,5 kg of print material total X axis went out of lubricant. Here what happened:
1) I turned on printer after not using it for at least 2 weeks
2) Ejected PLA filament
3) Loaded PET-G filament
4) Run print from USB drive
5) Printer failed to Mesh bed levelling
6) I cancelled print and selected print once again
7) Printer moved X axis by 1 cm, then started going down Z axis smashing into build plate. I reach power off while X axis was bend at least by 1 cm between left and right side.

After cleaning and lubing printer it printed out smoothly, but for some reason it totally ignored reading from Z-height probe and just smashed into the bed.

JohnnyDeer, 2020-05-19:
Hi @sergiusz-roszczyk , looks like you have PINDA issue, that could be reason for MBL fail and also for Z crash. Can you please describe how looked MBL fail?

sergiusz-roszczyk, 2020-05-19:
Hi @JohnnyDeer.

Printer was working (almost) perfectly fine on 4.0.1. My problems started after upgrade to 4.0.5 (mostly lack of confidence due to watchdog reboot - it may be related to issue with reading USB drive, I'm yet to switch from supplied USB drive to some other one). But back to MBL problems, there are two ways it fails (sometimes, second one is the one described above).
Scenario 1:
1. Power on
2. Immediately after Power ON select file to print from supplied USB drive
3. Printer tries to autohome but X axis moves only around 1 cm
4. MBL starts with correct height detection (no smashing on the print plate) but moves only around 3 - 4 cm left to right total between rightmost and leftmost side of X axis
5. Printer tries to print (which I cancel immediately for obvious reasons)
6. Power cycle printer, wait 1 - 2 minutes after boot, restart print - worked fine

Scenario 2: (this is what happened on my post above and probably lead to the crash - details on point 5)
1. Print head was parked near extruder
2. Printer did axis calibration by moving X axis to the right, then build plate to the back
3. Then moved down Z axis to the point on the build plate where it always did it
4. Started doing probing from the bottom row. First two rows went fine
5. On the third row I found that it moves only left and right within 3 - 4 cm and then "MBL failed" error message was displayed.

After that message I hit ok (or whatever it was showing) and started again. This time X was around 170 mm from the extruder.
1. Printer did (invalid) auto home by moving bed to the back (as usual) and X axis by 2 cm (fail here but no error message)
2. Started moving Z axis down and then smashed to the bed

I turned power off immediately and moved Z axis up by turning screw by hand and turned printer ON to let it cool down (I was going to print PET-G so 170 deg head, 85 deg bed - textured sheet).

After it cooled down I found that it is very hard to move X axis between lets say 30 and 150 mm from extruder. No obvious scratches, upper and lower rail look and measure parallel. So I opened up lubricant supplied with printer. Cleaned up upper and lower rail and lubricated them so X axis was going smoothly by hand.

Then I started printer again and selected autohome to check it out. It worked perfectly fine.
So I started again PET-G print. Two prints, two successes.

So I guess something got confused by stalled X-axis.
My printer was shipped 4.02.2020.

JohnnyDeer, 2020-05-20:
Thank you for information @sergiusz-roszczyk , it seem your issue is related to #421 . Did you tried factory reset after upgrade to 4.0.5? When you downgrade back to 4.0.4 it's not happening, right? Just in case, please make sure that X end part is perfectly perpendicular to bed.

sergiusz-roszczyk, 2020-05-20:
@JohnnyDeer yes, it seems very similar to #421. I haven't tried to downgrade to 4.0.4, but weird behavior with half movements ended after lubing the X axis.

I measured the X axis and it looks like there is around 1 - 1,5mm difference between center left and center right of the bed. I tried to loose or tight two screws that hold two pieces together but with no luck there. If you can point me to docs how to fully adjust it I'll try to do this tomorrow evening.

I did factory reset just a moment ago, went through full first time wizard perfectly fine.
I did first level calibration (-0,940 on textured sheet) and PET-G sticks fine. I'll try to print something.

Before doing factory reset I tried two more things that made software unhappy.
1) Power on
2) Calibration /  Auto home (it worked fine) (cold bed/head)
3) Calibration / Mesh Bed Leveling (cold bed/head)
Printer went through MBL just fine and then hung on "calculating results" screen. I had to power cycle it. Then I did factory reset mentioned above.

Cold = printer display 20 deg C.

JohnnyDeer, 2020-05-25:
@sergiusz-roszczyk if lubing helped, i´d like to consider it as solved. I meant, X axis perpendicular in direction like on photo, that could block bearings on axis. 

[image]

sergiusz-roszczyk, 2020-05-30:
@JohnnyDeer I've checked at it is. I placed PCB ruler on the bed and both rods are touching it. Maybe you can slip a sheet of 80g/m2 paper into one. I think it is perpendicular within measurement accuracy.
I also tried to unscrew and screw again those three screws that hold both part of printers together and Z-height difference on left and right side of the bed is smaller (within 1 mm).
I did few more prints this week and axis calibration worked fine.

So to sum up for others: If your printer have problem with calibrating X-axis or Y-axis - turn the printer off. Check if printhead and bed is moving smoothly by gently pushing them by hand. If not - check if they are perpendicular. If yes - check lubing using supplied lubricant.

JohnnyDeer, 2021-02-17:
@aurelw is this issue still relevant?
@sergiusz-roszczyk sorry for long pause and thank you for initiative and help :)

aurelw, 2021-02-17:
I guess as long as there is no real crash detection on the Z axis, this is still relevant as this would be the only real solution. 

Assuming a hard reset or crash during the print process is possible. This would result into a part being stuck to the nozzle and a subsequent homing on the Z axis would lead to a crash with possible hardware damage or bending of the buildplate.

closed 2024-07-04 by github-actions[bot]