prusa3d/Prusa-Firmware-Buddy #420: Menu behavior

geoper2, 2020-05-04:
When jumping from a submenu back to the main menu, for example from Filament back to Main, the highlight goes back to the first item which is Print.
The same happens when editing values inside a submenu. Upon exit the highlight goes to the first item on the submenu. This is furstrating because you need to scroll through all of the menu items to get back to the item you were before.

It would be better for the user experience to keep the highlight in the same place that you were before exit or back. When exiting or going back the highlight needs to remain in the same item and not move to the first item on the screen.

Crawlerin-SA, 2020-05-11:
If I may add, this would come in handy when browsing directories and subdirectories. Now it returns to the first directory (sorted by date of creation?). I believe if it returns to the directory you just were in instead of to the top, it would be a bit better experience.

danopernis, 2024-04-16:
I just checked and this is already solved. Not sure which version fixed that.