prusa3d/Prusa-Firmware-Buddy #421: Wierd Y axis problem

kdpainter, 2020-05-05:
I have noticed this issue recently with v4.0.5.  I do a print which completes successfully.  Then I load a the gcode file again and it does the mesh bed level and starts the print but it gets stuck at the middle of the bed, make a shudder sound and starts the print in between the middle and the front.  My object will fit in that area so I let it go and it printed fine.  However, if I stop the print and hit the reset button, the print (with the same gcode) prints perfectly.  There is nothing binding the printer in any way.  That makes me think there is a memory corruption issue.  This just showed up with v4.0.5.

JohnnyDeer, 2020-05-05:
Hi @kdpainter , thank you for reporting. Did it happen to you just once, or it's repetitive problem? Could you please attach Gcode relating to this issue?

kdpainter, 2020-05-05:
Hi @JohnnyDeer ,  It has happened twice.  The first time, I cleared it by hitting the reset and it printed fine.  The second time, I let it print in the front half of the print bed.  It wasn't exactly the same file but had minor tweaks to dimensions in my CAD program.  I sliced it in Simplify3D with a profile that has been working perfectly.

I am not sure how to attach a file.  It is 5MB so probably too big.  I uploaded it here:
https://www.dropbox.com/sh/ai4kd57cwy1myb6/AACT7Bp5VwrbjeVKsYP9Q3wla?dl=0

Thanks!

JohnnyDeer, 2020-05-06:
@kdpainter thank you for info, I´ll try to reproduce it. Just one more question,  after first print you just clicked on reprint, right?

kdpainter, 2020-05-06:
@JohnnyDeer No, let me be more clear as to what I did.  The first time, it hung up at the middle of the bed.  This was during the mesh level routine.  I hit the reset and then printed the same gcode file.  It went fine that time.  The second time this happened, it was with a different gcode file.  Same object but with dimension adjustments, same slicer and profile.  That time, it hung on the mesh level routine at the center of the bed but I just let it print.  It printed fine just in the front half of the bed.  I have since seen it happen again but this time the printer had just freshly booted.  A reset cleared the problem and the same gcode file printed fine.

This always occurs in the middle of the mesh level routine and the stepper sounds like the y-axis was being restrained but nothing is restraining it.  I had not seen this issue until I flashed v4.0.5 firmware.

Crawlerin-SA, 2020-05-08:
Greetings, I have encountered very similar issue, twice now, both after I hit Reprint. After reaching 170/85 degrees (PETG), the bed does not move to the front and printer probes 16x going left and right at the end of Y axis, motor crashing when it tries to move back but cannot. After that finishes, it homes, increases temp and prints normally, albeit whole print is shifted by like 0.5 cm to the back including priming line. To me it seems like it somehow skips or ignores the "G28 ; home all without mesh bed level" line just before G29 in starter Gcode. 
After this print finished and I hit reprint (without restart), it prints normally. Restart solves this too.

Crawlerin-SA, 2020-05-08:
Here's the gcode I used. It's just few objects to deplete roll.
PrusaSlicer
Version:   2.2.0+win64
Build:     PrusaSlicer-2.2.0+win64-202003211145
[pozicer-orig21_0.2mm_PETG_MINI_47m.gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4596722/pozicer-orig21_0.2mm_PETG_MINI_47m.gcode.zip)

JohnnyDeer, 2020-05-12:
@kdpainter @Crawlerin-SA thanks for detailed information. Unfortunately i'm not able to reproduce it. If you´ll notice some more context around this issue, please leave comment here.

kdpainter, 2020-05-12:
@JohnnyDeer Thanks for looking into it.  It seems very intermittent.  I have not seen the issue since my last post.  It does seem that it only occurs during the mesh bed leveling and that was one of the changes in v4.0.5.

Crawlerin-SA, 2020-05-12:
Me neither, I even take out phone to record video of it in case it happens every time I start printing, but nothing as of yet. @JohnnyDeer is there anything we can do in case it occurs? Save memory dumps or something? There is an option to dump crash in Settings but this does not seem like crash, printer continues to work normally albeit with wrong presentation of its print area.

JohnnyDeer, 2020-05-13:
@Crawlerin-SA as you wrote, its not crash... so maybe log from OctoPrint could have some information about this behavior. But that means always print trough OP and wait till it happen. Hard to find out issue wich is not reproducive. Anyway thanks for your help :)

kdpainter, 2020-05-15:
@JohnnyDeer My problem came back to where resetting no longer worked.  Neither did power off/on.  It seemed like it found where home was incorrectly during mesh level.  I needed the printer to work so I downgraded the FW to 4.0.4.  It seems to be working normally again but I don't have much print time yet.   One thing I noticed, I had to do a factory reset as part of the process.  I don't think I was prompted to do that when going from 4.0.4 to 4.0.5.  @Crawlerin-SA that may be something to try.  Do a factory reset with 4.0.5.

kdpainter, 2020-05-17:
@JohnnyDeer  Just following up after doing quite a few prints.  After my downgrade to 4.0.4, I have not seen any of the behavior I saw with 4.0.5.  Prior to the downgrade, the printer was homing to the back of the print bed.  It would go back as far as it could and then start doing the mesh level from there.  A power cycle and restarting the print would yield the same result.  At this point, the printer was non-functional.  I did the downgrade by putting 4.0.4 on the thumbdrive and double-pressing the wheel when it was loading up and then flashed.  I am wondering if not doing a factory reset after the install of 4.0.5 could cause this problem?  Previously sliced files that were failing on 4.0.5 printed fine on 4.0.4  after doing the factory reset and re-calibration.  

It is working well on the previous version of firmware so I will stay on that for now.  I am making respirators for my family for now so going to keep on going with that version.

JohnnyDeer, 2020-05-20:
@kdpainter we are working on find out whats wrong. Thank you for all information.

kdpainter, 2020-05-26:
@Crawlerin-SA any news on if the factory reset solved your problem?  I now started having watchdog timer resets at the beginning of every print regardless of gcode.  Debating on what to do now.

JohnnyDeer, 2020-05-27:
@kdpainter could you please try to make and upload video of watchdog reset on begin of print?

Crawlerin-SA, 2020-05-27:
> @Crawlerin-SA any news on if the factory reset solved your problem? I now started having watchdog timer resets at the beginning of every print regardless of gcode. Debating on what to do now.

Well, the "problem" here is that I haven't encountered this issue again. Granted I don't print as much as I used to i.e. hours and hours a day (as I stopped printing mask accessories), but other prints I did all was OK. Printer homed and probed normally.

kdpainter, 2020-05-27:
@JohnnyDeer It seems that this is actually a hardware issue.  when it is starting to warm up the bed, the indicated bed temp drops to -15C and then I get a reset and when it comes back up, it says watchdog reset.  If I power cycle at that point before it cools down, it reboots and still says -15C.  I suspect the thermocouple is shot.  Frustrating as this is the second hardware problem I have had with this new printer.  I brought my Mk3 back from my other place.  It needs a new controller board as my dog broke it.  I have the controller.  Hopefully support will get back to me on the thermocouple.

@Crawlerin-SA Glad you are not having issues.

JohnnyDeer, 2020-05-28:
Summarizing this issue:
Weird axis movements are not related to firmware, but hardware. Probably bearings jamming or not enough lube.
Watchdog resetting at beginning of print is caused by damaged bed thermistor. 
In both cases will help you our support.

Can we consider this issue as solved and close it?

kdpainter, 2020-05-28:
@JohnnyDeer I would not say the weird Y-axis issue is not related to firmware.  That was solved for me by downgrading to 4.0.4.  I did many prints with it with no issues.  What is unknown is if the factory reset cleared something out.  If I updated to 4.0.5 and then did a factory reset, perhaps that would have some positive effect?

The watchdog reset issue is hardware and seems to be a faulty thermistor.  Unfortunately, my printer is now a brick so I can't test anything.  I have not heard from support.

Crawlerin-SA, 2020-05-28:
It may be an explanation. that it was more than usual resistance with initial movement, making motor detect crash and printer "thinking" it's at the beginning of Y axis. I try to keep rods and bearings clean and lubricated and printer is in enclosure to avoid dust, but cannot exclude this is what happened.

The probing routine did not raise any error and that one was really at the end of axis with no way to move further :-)

JohnnyDeer, 2020-06-09:
@kdpainter unfortunately we are not able to reproduce it, that means we cannot find out if is something wrong in FW. I´m afraid i cant help you at this monet. If you find out how to reproduce this behavior, please create new issue with all needed information. I´m closing this issue, thank you for understanding.

closed 2020-06-09 by JohnnyDeer

JohnnyDeer, 2021-05-17:
@kdpainter have you tried any new version of FW? Do you still have this issue on them?

kdpainter, 2021-05-17:
@JohnnyDeer My Prusa Mini has not worked for a long time.  The bed heat sensor failed and support did not have a replacement.  I have not tried to talk to them again about it.  It was still under warranty at the time but now it isn't.  I had another mechanical issue with the printer which is a indicator that it is a bad idea to pre-order things in general.