prusa3d/Prusa-Firmware-Buddy #431: Mesh Bed Leveling Issue

nilsstreedain, 2020-05-10:
I've been using version 4.0.5 daily since it's release with no problems, until today. When I initiate Mesh Bed Leveling, or the machine begins mesh bed leveling before a print; the machine, while measuring the 8 stops on the left, only moves upwards away from the bed and not words it. This is causing catastrophic results when it comes to printing. I've attached a compressed video of the issue: [Untitled Project.m4v.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4605006/Untitled.Project.m4v.zip)

I've tried reflashing the most recent version of firmware, can seem to flash older versions. I've also set it back to factory defaults but the issue still persists.

bobc, 2020-05-10:
This is the well known fault with the MINDA sensor, it is not firmware related. The wiring develops an intermittant break, when it is stretched to the left side, the probing fails.
You need to contact Prusa support and get a new MINDA sensor. If you are confident you could attempt to repair the cable.
Also consider the "MINDA saver" mods which attempt to alleviate stress on the cable loom.

JohnnyDeer, 2020-05-12:
Hi @nilsstreedain , please report your problem on chat, our support will help you ( https://shop.prusa3d.com/en/  - chat icon is in lower right corner )

closed 2020-05-12 by JohnnyDeer