prusa3d/Prusa-Firmware-Buddy #439: Cooldown timer setting after filament change/unload

Fliens, 2020-05-12:
When changing/unloading filament the temperature stays for a certain amount of time at the changing temp. My suggestion ist that there could be a setting for this cooldown so that you could set the seconds for how long the temperature stays at the changing temp. And if you set it to 0 it immediately cools down after changing the filament :)

JohnnyDeer, 2020-05-13:
Hi @Fliens , this issue is related to #221 , thank you for suggestion.

scubagregory, 2020-06-09:
If the mini would just work like the MK3S that would be better.  It stays hot longer and also "beeps" notifying the user of needed attention

JohnnyDeer, 2020-06-24:
Hi @Fliens , filament loading function have changed behavior in FW 4.1.0-RC1. Filament is loaded to gears immediately, before reaching temperatures, and will be done automatically after their reaching. Thorugh this change is coldown issue avoided. I´m closing thos issue as solved.

closed 2020-06-24 by JohnnyDeer