prusa3d/Prusa-Firmware-Buddy #441: Gcode preview in file explorer

Fliens, 2020-05-12:
Since the name of the file is not always describing it perfectly esspecially when there are designs bodies combined in one gcode, a small image of the gcode next to the filename would be awesome.

This would just simplify the way you find and choose files and how you use the file explorer :)

 (Feature Request)

jkavalik, 2020-05-17:
@Fliens check https://github.com/jkavalik/GcodeThumbnailExtension (disclaimer: I did it and I am NOT affiliated with Prusa)
I tested it on Windows 8 and 10.

Fliens, 2020-05-17:
@jkavalik that's pretty great but my request was for the prusa mini file explorer :)

jkavalik, 2020-05-17:
@Fliens ah, I see now, sorry then.. Yes, that woud be useful. Or maybe have a <previous>/<next> in the preprint preview that shows the thumbnail already to browse files with details.

Eldenroot, 2022-11-26:
I think that you can see the gcode preview when you open it in the printer... this should be enough, the image is large and you can see what you need.

@Prusa-Support

Prusa-Support, 2022-12-10:
The current enlarged preview image (upon clicking) can be satisfactory in most cases, but that is not the object of this feature request I believe.
The request is to have an image right next to the filename in the list - without selecting the file - if I understand correctly.

Michele Moramarco
Prusa Research

Eldenroot, 2022-12-10:
It could save some time to find the right file

Now the text scrolling speed for longer file names is really slow... this could be really helpful

closed 2024-07-04 by github-actions[bot]