prusa3d/Prusa-Firmware-Buddy #443: Preheat setting issue

abeq9493, 2020-05-13:
When ever I reset my preheat from any pre-settings of the menu it never changes to the pre-settings when I try to purge. It stays on the Flex pre-settings of 240C. 

The same thing happens when I try doing a first layer test. It stays on the Flex pre-settings of 240C when I try to  set it to anything other than Flex pre-settings. It levels the bed when I do a first layer test at the Flex pre-settings temps also. it doesn't drop to 170C.

JohnnyDeer, 2020-05-13:
Hi @abeq9493 , when you load filament, in this case flex, printer remember that and always heat up to flex, if you want to purge or unload. If filament is not loaded, printer will alwas ask you wich temperature to youse when you want to purge.
Second part of issue is relataed to #157 .

abeq9493, 2020-05-13:
Doesn’t matter which I select. It still faults back to flex. It won’t change to what I change it to. No matter which I select it stays on flex presets.

------ Original Message ------

From: JohnnyDeer
To: prusa3d/Prusa-Firmware-Buddy
Cc: Abe A Quintana, Mention
Sent: May 13, 2020 at 2:44 AM
Subject: Re: [prusa3d/Prusa-Firmware-Buddy] Preheat setting issue (#443)

Hi @abeq9493 (https://github.com/abeq9493) , when you load filament, in this case flex, printer remember that and always heat up to flex, if you want to purge or unload. If filament is not loaded, printer will alwas ask you wich temperature to youse when you want to purge.
Second part of issue is relataed to #157 (https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/157) .

JohnnyDeer, 2020-05-14:
@abeq9493 ok, do you tried factory reset? If no, please try it.

FEsmondeWhite, 2020-10-08:
I have a similar issue. The two MINIs that I have always preheat to 170C, regardless of what I select. For example: selecting preheat for PETG, the display shows nozzle temp stabilizing as 170/230 C. Then the nozzle temp flashes, while staying at 170 C. This is with firmware 4.2.1-final-+2072, bootloader 1.0.0 buddy board 1.0.0. Is this expected behavior? I have loaded PETG into the printer, and it still runs at 170 C. 

Maybe this is a result of upgrading from original firmware to latest version?

I think it's because of the default filament preheat temp in " Prusa-Firmware-Buddy/src/common/filament.h "

`**static const float PREHEAT_TEMP = 170.f;**`

jkavalik, 2020-10-08:
@FEsmondeWhite Thats because 170°C is the temperature for bed leveling, so if you start a print, it does probing at 170 and then heats up to 230. Before this change preheat heated to 230 (in case of PETG) and then when you started the print it had to wait while hotend cooled down to 170.

FEsmondeWhite, 2020-10-08:
This makes it very inconvenient to clean the nozzle before printing. I
appreciate the reduced delay in bed probing, but I still need to ensure the
nozzle is clean before prints, and this requires the filament print
temperature...

The current behavior doesn't do what is explicitly called out on the user
interface. When I click preheat to 230, and the display shows the setpoint
at 230, it doesn't make sense to have an actual hot end setpoint of 170.

If someone just wants to print, they can just start a print with the nozzle
starting from cold.

Maybe this would be less problematic if there was a firmware calibration
item for nozzle cleaning? Like park with a high z level (120+mm), and heat
the nozzle to the filament print temperature.

The bed heating is the time limiting factor anyways.

JohnnyDeer, 2021-02-17:
@abeq9493 is this issue still relevant?
@FEsmondeWhite preheating to 170°C is not related to main issue, please make feature request if you have ideas how to improve this behavior, thank you.

JohnnyDeer, 2021-05-13:
I'm closing this issue because of no response.

closed 2021-05-13 by JohnnyDeer

OJFord, 2021-06-09:
@JohnnyDeer Please re-open, I was just about to open a duplicate before I found this.

The issue and use case I have is exactly as described by @FEsmondeWhite:

- I want to be able to clean the nozzle of PETG before starting a print;
- I select preheat > PETG
- It heats as far as 170/230 nozzle & 80/85 bed
- It hangs there forever
- I can't clean the PETG off at this temperature

Annoying workaround is to start the print, pause, and then scrub before it cools down too much (because pausing also won't hold that 230 temp).