prusa3d/Prusa-Firmware-Buddy #460: Feature request: webcam support

martin8425, 2020-05-15:
When the Mini was introduced Jo talked about webcam support in an interview. Is that still coming? I would love to be able to use the Prusa interface for remote monitoring with a webcam like you can do with Octoprint ;)

JohnnyDeer, 2020-05-18:
Hi @martin8425 , this feature is on our to do list :)

martin8425, 2021-06-11:
@JohnnyDeer any news on this? 😊

mtn44224, 2022-05-14:
I'm quite sad, that this feature seems to be cancelled entirely :(

JustAnother1, 2022-05-15:
The STM32 micro controller used in the mini is probably not powerful enough to allow this feature. Sending a picture once in a while should be possible even for the STM32.

Does anybody know a good USB-cam with open source drivers, that is widely available for cheap?

Prusa-Support, 2022-05-27:
Hello! Thank you for your comments and sorry for intervening so late on this thread. Unfortunately, we don't have info yet for this aspect, I also wouldn't know about the possibilities for the STM32 - I brought these aspects to our devs attention, as soon as there will be more info I'll update in this thread. 

Alessandro Pantaleo
Prusa Research

pagoru, 2023-08-19:
No more updates...

pagoru, 2024-06-13:
Please, disable the github bot, It's stupid to bump a feature that everyone would be benefied...

Prusa-Support, 2024-06-18:
With nearly 4000 issues listed over the years - among opened and closed ones, hundreds of which are still open, many of which may be stale or invalid - _the bot is especially useful for drawing attention to valid issues like this one_.
We appreciate your collaboration in keeping this issue open.

For now, please consider the already available alternative options.
https://help.prusa3d.com/guide/camera-setup-for-prusalink-prusa-connect_470943
Moreover, thanks to the Prusa Connect Camera API, custom applications are possible too, and the project PrusaConnect ESP32-CAM mentioned in the link above is a great example of it.
https://help.prusa3d.com/article/prusa-connect-camera-api_569012

Michele Moramarco
Prusa Research

closed 2025-09-27 by Prusa-Support