prusa3d/Prusa-Firmware-Buddy #465: Sometimes list of files on the USB is incomplete

jkavalik, 2020-05-17:
It happened to me multiple times and I think it depends on the number of files on the thumbdrive, but I did not specifically debug any numbers.

When I put a new file on the drive and insert it into the printer, I am offered to print that new file. It works fine. But if I return to the main menu and go to the print dialog, the file is not present in the list. If I pull the usb out and reinsert it again I get the dialog to print that file directly.

If I delete some older files from the drive and insert it into the printer again with the "problematic" file untouched, it shows in the list again.

JohnnyDeer, 2020-05-18:
Hi @jkavalik, did you tried different usb devices, is this happening just with one? Could you please upload file with which is this happening?

jkavalik, 2020-05-18:
@JohnnyDeer Hello, I just tried to enforce it:
- checked that it looks fine
- duplicated the files on the drive
- sliced a new gcode file

Results were that on insert it suggested the newest file for print but when cancelled I was not able to find that file in the list. I copied the same (complete) content onto another drive and the results were similar, only the order changed because the copying changed creation date for the files.

So I counted the items:
- original drive - 61 gcode files, 2 directories and '..'
- second drive - 62 gcode files, 1 directory and '..'

In total 64 entries in the list.
The drives contain 2(first) or 3(second) directories 135 gcode files, lan_settings.ini and firmware file.

JohnnyDeer, 2020-05-19:
I did reproduced it and can confirm bug. Thank you for reporting and info @jkavalik

JohnnyDeer, 2020-05-21:
Hi @jkavalik , updated information, this issue will be fixed in next FW version.

JohnnyDeer, 2020-06-24:
Hi @jkavalik , this bug is solved on FW 4.1.0-RC1, please try it out :) . I´m closing this issue as solved.

closed 2020-06-24 by JohnnyDeer