prusa3d/Prusa-Firmware-Buddy #466: USB not detected

Fliens, 2020-05-17:
[image]
As you can see in the photo, the usb is in the printer (upper right corner) but I can't select "print". I put the usb in the printer right after the startup was finished.

BUG

0xygen8, 2020-05-17:
USB2 or USB3 stick the mini has some issues with some USB3 sticks and the original PR has sometimes some bad block issues resulting in exception errors (mostly while printing) my advis take an old USB2 stick (black inside on the plus often indicates that is USB2 --while USB3 sticks have a blue inside in the plug) and try again

Fliens, 2020-05-17:
@0xygen8 I just took it out and back in and it worked again. But I was wondering because it's the original usb stick that got shipped with the printer

0xygen8, 2020-05-17:
I had the same issue with a USB3 stick now I only print with 'old' USB2 sticks that I had laying around -- you cloud try to reformat (not quick) the original USB stick and see if that helps

JohnnyDeer, 2020-05-21:
Hi @Fliens , may be faulty USB stick. Did you tried to reformat it?

Fliens, 2020-05-21:
@JohnnyDeer it worked again after taking the stick out and putting it back in again :)

JohnnyDeer, 2020-05-25:
@Fliens it happend to you only once? If yes, it's probably USB stick problem.

icdvdt, 2020-05-29:
this happens consistently on first power-up and if I press 'reset' then it typically mounts the drive OK.

Dronitoni, 2020-06-01:
same situation here. The MINI doesn't accept any USB-Stick even if they are formatted with win7 or win10. 
I also tried SD-cards, USB-extension cable........

My workround is to reboot the MINI with the inserted USB-Stick every time I want to start a new print.

JohnnyDeer, 2020-06-02:
@Dronitoni thank you for detailed information. Next FW version will have better sensitivity to recognize usb stick. This problems should be solve.

zoltan-l, 2020-06-08:
Gus ,can you confirm or deny, that your print files are not in root directory but somewhere in any subdirectory? I could replicate the issu: print files not in root directory - removing and inserting Flash disk during operations affects only thumb icon to (dis) appear on the screen not anabling the print Icon, booting anabling the print Icon - known issue and should be corrected by 4.1.0.
Work around : put at least one gcode file in root directory.

Dronitoni, 2020-06-08:
In my case, I have gcode in the root directory as well as in subdirectories on the USB stick.
In all cases it doesn't work. 
After reboot I  have access to root and subdirectories.

Chleba, 2020-06-10:
Hi,
Thank you for reporting this bug. It is already fixed and will be implemented in next firmware version.
Reason why it's happening is for not having any valid gcode file (*.g or *.gcode) in root directory so QUICK PRINT feature does not happen and screen is not re-rendered.

JohnnyDeer, 2020-06-23:
Hi @Fliens @Dronitoni , FW 4.1.0-RC1 is out now and should solve this issue. Please try it out.

Dronitoni, 2020-06-23:
Yes, it works perfect now.

Thank you.

JohnnyDeer, 2020-06-24:
Great, thank you all for help. I'm closing this issue as solved.

closed 2020-06-24 by JohnnyDeer