prusa3d/Prusa-Firmware-Buddy #480: [BUG] One-Click-Printing ignores subdirectories

Saij, 2020-05-19:
**Printer type** Prusa MINI
**Printer firmware version** 4.0.5

**Describe the bug**
When I save a new G-Code file inside a subdirectory on the USB drive, One-Click-Printing shows only the newest file on the root directory.
  
**To Reproduce**
Create a G-Code file inside the root directory. After that, create a subdirectory and create some G-Code files inside this directory.
  
**Expected behavior**
Maybe an option to deactivate the One-Click-Printing function or a scan inside subdirectories would be the way to go in my opionen.

JohnnyDeer, 2020-05-21:
Hi @Saij , this behavior is on purpose, because it may take long time to scan all subfiles and files, depended on their amount. We can transfer this issue to feature request.

Saij, 2020-05-21:
I think the transfer would be nice because I love to organise my multi part prints in subdirectories ^^

hawwwran, 2020-06-24:
I would be happy enough if long hold on PRINT button in menu would scanned and offered the last modified file.

closed 2024-07-04 by github-actions[bot]