prusa3d/Prusa-Firmware-Buddy #489: Filament runout/change procedure does not work

dwineman, 2020-05-23:
Filament ran out during a long print, tripping the sensor. The printer automatically unloaded, then went to "Push filament into extruder - continue". I got a new spool ready, pushed it in, and pressed the knob. But instead of loading, it exited back to the Tune menu and stopped heating. I tried several times to resume the loading procedure but nothing worked, and eventually it crashed out to the "Requested temp not reached - reset printer" screen.

I would very much like to see this fixed as it makes long prints impractical on this printer.

JohnnyDeer, 2020-05-25:
Hi @dwineman , thank you for reporting. Which FW version is on your printer? Is this happening always, or after update FW?

dwineman, 2020-05-25:
I have FW version 4.0.5, and I'm printing via OctoPrint. This was the first time I've had a filament runout, but I will experiment and let you know if it happens consistently.

JohnnyDeer, 2020-05-26:
Issue related to octoprint, that is important information. Then this is duplicate of #362 and will be closed.

closed 2020-05-26 by JohnnyDeer

bassco, 2020-05-27:
@JohnnyDeer 

I have FW version 4.0.5 on the MINI and using PrusaSlicer 2.2.0 - my son pressed Pause during a print, bed stayed on, extruder temp dropped and we had to abandon the print. There was no way to continue the print.

Not using Octoprint.