prusa3d/Prusa-Firmware-Buddy #491: X axis homing, while website is open, belt skips

Gisi0, 2020-05-24:
When the website is open, homing of x axis not working well. I think, he is moving until a timeout. Belt is skipping.
When he stops, x axis homing is successful and he is going further.
I don't know if it is every time.
When the website is not open, no problems.

JohnnyDeer, 2020-05-25:
Hi @Gisi0 , could you please describe steps needed to reproduce this issue? Which FW version is on your printer?

Gisi0, 2020-05-25:
I will try to make a video.
Firmware is 4.0.5

JohnnyDeer, 2020-06-09:
Could i please ask you for more information, and mentioned video @Gisi0 ? Thank you.

JohnnyDeer, 2020-06-15:
Thank you @Gisi0 , then i'll close this issue. If you find out how to reproduce it, or you will create video, we can reopen it.

closed 2020-06-15 by JohnnyDeer