prusa3d/Prusa-Firmware-Buddy #493: ABS as Filament Option missing

feutl, 2020-05-25:
The problem got raised in the forum
https://forum.prusaprinters.org/forum/hardware-firmware-and-software-help/abs-profile-on-mini

There is no "setting" for ABS as a filament to change, load or unload within the Prusa Mini firmware. This is a little confusing and the temperatures which should be used for ASA and ABS also seem to be different as I have understood.

Can this profile can be added somehow to the firmware?

Thanks

JohnnyDeer, 2020-06-09:
Hi @feutl , we don't have plans to implement ABS to FW at this moment. ASA preheat can be used for load ABS, yes recommended temperatures can be different by +-5 °C, but that is recommendation  mainly for print. Besides, real temperature on hotend have range in wich can be different from setted temperature, ... if is printer preheated to 255°C, is possible to measure +-5°C on hotend. So dont worry to use ASA preheat for ABS load.

rtyr, 2020-06-21:
Actually, plans have changed, ABS option will be implemented. It is already implemented in 4.1.0-RC1 along with some other materials.

JohnnyDeer, 2020-06-22:
I can confirm, ABS preheat is in latest FW version. I'm closing this issue as solved.

closed 2020-06-22 by JohnnyDeer