prusa3d/Prusa-Firmware-Buddy #511: Can't print from Nested Folders

jweaver100, 2020-05-28:
This is so obvious, I suspect its already been raised, but just incase I figured i should do it. This may even be design intent.. So feel free to close this if its a known or non-issue.

I am on the latest 4.0.5 Mini firmware and have been cleaning up my USB sticks.

I have added a folder inside a folder for some of my common prints..

So as an example I have files under:

/regular prints/bullets 

I tried to print one of the GCODES in this folder and it just heated the hotend and bed.. and then sat there.. I was unable to stop the print, so I pressed the reset and tried again.. This time the heaters turned off.. Everything cooled... And again nothing.

I then moved the "bullets" folder into root.. And all is working normally again.. So it seems that the Mini does not allow 'nested' folders'.

icdvdt, 2020-05-29:
I have this exact problem as well...

JohnnyDeer, 2020-06-01:
Hi @jweaver100 , this is known issue and should be solved in next version of FW.

hawwwran, 2020-06-16:
I have very similar, but different problem. My problem is the nozzle and bed stay cold. Because (as the display shows) it expects to heat "up" to 0°C! This bug is unbelievable to be honest.

DRracer, 2020-06-23:
@hawwwran @icdvdt @jweaver100 could you please try upgrading to the latest 4.1.0-RC1 and report back to verify the if issue still exists in the lastest release? We did quite big changes to solve problems like this.

hawwwran, 2020-06-23:
I can confirm the bugs I reported are fixed.
- Printing from deeper tree structure works as expected
- USB stick is correctly recognized if inserted when printer is running (and there is deeper tree on the stick)

JohnnyDeer, 2020-06-24:
Great, then i'm closing this issue as solved.

closed 2020-06-24 by JohnnyDeer