prusa3d/Prusa-Firmware-Buddy #518: Length of purge when loading filament

wengeol2, 2020-05-31:
On the Prusa Mini (4.0.5) the initial purge of filament is a bit short when loading fresh filament. when selecting "NO" that the filament doesn't have the correct color, the second purge is too long.
Would it be possible to make the first purge longer or make the second purge much shorter?

okaerin, 2020-05-31:
What about a manual override of the second purge, so it could be stopped earlier?

JohnnyDeer, 2020-06-04:
Hi @wengeol2 , at this time is longer second purge on purpose. Scenario of loading and purging filament will be changed in next FW version.

Texy, 2020-06-11:
One problem is that if the filament doesn't catch on the first attempt, it is really painful to manually keep saying 'No' to the right colour selection in order for the filament to be loaded into the extruder. Hopefully the new firmware will address this ?

JohnnyDeer, 2020-06-15:
@Texy yes, this will be also improved in version 4.1.0. :)

Texy, 2020-06-15:
Thanks - the next version is a long time coming ;-) . Any Beta or RC release available ?

nsaubi, 2020-09-16:
Any news on this ? I just received my mini prusa, installed fresh 4.2.1 fw, and each time I want to change color, being at printer start or during a print, it takes me 1 push to remove the filament (ok), but like 5+ pushed on "no" to purge the new filament out.
Maybe adding a manual purge option there would be cool, like the live adjustment of the z axis (with the wheel), where turning any side would just purge the filament.

x10102, 2020-09-20:
Well you can use the wheel to push filament into the extruder by going to
Settings > Move Axis > Extruder

ShadowDrakken, 2022-06-14:
I'm having the reverse issue. The purge is more than twice as long as I need for filament changes in the MINI+.
It'd be nice to have access to the purge length in the FW settings menu

Prusa-Support, 2022-06-21:
Hello! Sorry for the issue and for the late response from my side. All noted and I'll bring this to our devs attention, so that the improved feature can be evaluated - sadly, no guarantees anything would be implemented for now but we'll absolutely evaluate it. 

Alessandro Pantaleo
Prusa Research

ulab, 2023-05-09:
I second an option to be changeable in the settings menu. I usually have to purge three or four times for the color to be correct. Being able to configure a longer purge, that can still be stopped with a click would make this a lot simpler.

hackwin, 2024-02-18:
I would also like a feature to change the purge amount by a multiplier when loading filament.  It's a waste of time having to wait for a little bit of plastic to purge, and see the color is still mixed, and have to repeat 2 or 3 times.

closed 2024-07-04 by github-actions[bot]

DarKOrange75, 2025-08-07:
5 years (!)  later and still no fix, I must click 3 times the more button to get the filament out of the nozzle...

ulab, 2025-08-07:
Yes, that's what "Closed as not planned" means.

cnaw90, 2025-08-29:
For those still interested in this request, I just opened a new one: #4784. Please go comment and support it!