prusa3d/Prusa-Firmware-Buddy #535: [FEATURE REQUEST] [Mini] Preheat to Probing Temp (on Startup)

avrcore, 2020-06-05:
Preheat (170/60°C) in the Menu or optional on Startup. This is a easy workaround for the first layer calibration Temperature dependency. 

Thank you Guys!

jkavalik, 2020-06-06:
Looks like this is already in the works - #321

avrcore, 2020-06-06:
Oh, didn’t see that. But The Idea of Preheat on startup / power-on is still new.

JustAnother1, 2020-06-07:
Wouldn't heating on power on be an safety issue in case of (longer) power outages ? (Just asking, should be considered before implementing this)

avrcore, 2020-06-07:
1. Should be optional. The Mini just cant be used from a cold-start. 
2. This would straight up mean heating in general is a safety issue (which is not).
3. The SOC can restore default settings on power loss or critical events.
4. Doesnt make sense for me. A power outage would restore the the state i want anyways. Cold -> outage -> Cold. Or Warm -> outage -> Warm. But never Cold -> Warm, which would be undesired.

JustAnother1, 2020-06-07:
@avrcore The Heaters are the things that burn down the house if something goes wrong. And yes that already happened also more than once.

I just wanted to make people aware that there are safety concerns with heating.

Making this feature a default off one will already help.

JohnnyDeer, 2020-06-16:
Hi @avrcore  , this feature request cannot be done because of safety. I'll close this issue as invalid. Thank you for understanding.

closed 2020-06-16 by JohnnyDeer