prusa3d/Prusa-Firmware-Buddy #550: [BUG] Crashes into print after filament change

x10102, 2020-06-09:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** - MINI

**Printer firmware version**-  4.0.5

**Original or Custom firmware** Original

**Optional upgrades** - Filament Sensor

**USB drive or USB/Octoprint**
The included USB drive.
  
**Describe the bug**
  After about two hours of printing the sensor detected the filament running out and 
  the printer paused. I swapped the filament for another roll (Both PLA). The leftover 
  filament was unloaded and the new one loaded just fine. When the printer started 
  printing again the Z axis was lower than before and the head crashed into the print, 
  destroying it.  

**How to reproduce**
  1. Start a print with insufficient filament.
  2. Let it print until the filament runs out and the sensor detects it.
  3. Change the filament according to the instructions on screen.
  4. Wait until the printer resumes.

**Expected behavior**
  The printer resumes printing where it stopped with the new filament.

**G-code**
  [gcode.zip](https://github.com/prusa3d/Prusa-Firmware-Buddy/files/4754213/gcode.zip)

**Crash dump file**
  [dump.zip](https://github.com/prusa3d/Prusa-Firmware-
Buddy/files/4754209/dump.zip)

**Video**
  Don't have one.

JohnnyDeer, 2020-06-16:
Hi @x10102 , unfortunately i'm not able to reproduce it. If you could make video, it will helps. Thank you

x10102, 2020-06-18:
@JohnnyDeer Weird, it didn't happen again after that. Guess it was just some one-time glitch

closed 2020-06-18 by x10102