prusa3d/Prusa-Firmware-Buddy #561: [BUG]First Mesh bed levelling attempt fails everytime

jsilva-iii, 2020-06-10:
**Printer type** - [MINI]

**Printer firmware version**-  4.0.5, .

**Original or Custom firmware**  - Original

**Optional upgrades** -  Filament Runout Sensor

Printing from :**USB drive 

After powering on printer, load filament, selecting print, printer cools temp for mesh bed leveling, it homes, to the front right, clicks  four times against the right hand end ( as viewed from front), moves bed forward, repeats, moves bed forward, print head then probes the remaining eight points properly. I usually cxl the print and restart after which the mesh bed levelling routine works as expected, i.e. homes, then probes all 16 points while moving around the bed

How to duplicate:
Turn on printer and attempt to print, first attempt at mesh bed leveling will do the same.

So far I have printed different 6 items all of them behave in exactly the same way.

Video only has the second row the first four points the head is completely against the right hand side, the leveling for the first four points is done in the same spot

I can email the video file as it's too large to upload

JohnnyDeer, 2020-06-16:
Hi @jsilva-iii , thank you for reporting. Please send video to honza.jelinek.tester@gmail.com , it will helps a lot :)

jsilva-iii, 2020-06-16:
Johnny

Hi, this happens almost all the time, after resetting it prints perfectly.

https://photos.app.goo.gl/dvuZxLrutRPe1g3L7

Thanks
Jorge

derenma, 2020-06-17:
Looks like a faulty stepper driver as the X carriage should not be slamming against the end of the rail.

monsterguy, 2020-06-18:
I have the same issue with bed levelling malfunctioning on the first attempt after powering on.
It heats to 170/60 and then does the following - 
https://photos.app.goo.gl/CP9npE8aaZeGhF2h8
The video ends with me cancelling the print as it starts to heat towards print temperature.

All subsequent prints until power off operate as expected.

[Version info - https://photos.app.goo.gl/J9rQVZP5TvKrfF756 ]

jsilva-iii, 2020-06-18:
Hi - just a correction - I don't reset the printer, I cancel the print and restart it.

4nthonylin, 2020-06-26:
Just wanted to chime in, I also see this issue and a reset always fixes it as well.

posttenebre, 2020-07-20:
I have the same issue. It does not occur when the print head is more distal from the Z screw side. So if you change the end gcode for the printer to position the head at  X2 it will not have issues with mesh bed leveling on the next start up. 

`G1 X2 Y180 F4200 ; park print head (previous x178)`

It has not been an issue since firmware 4.1.0-final, but I figure I would put my hacked solution into the comments here in case someone else needs a quick solution.

JohnnyDeer, 2020-11-09:
Hi, is this still a issue?

jsilva-iii, 2020-11-09:
Yes, I just move the head to the middle of the arm and I don't have that
problem

monsterguy, 2020-11-09:
Yes, it still happens every time unless I use one of the methods to avoid it.
Firmware and slicer updates have not changed or fixed the issue.

jsilva-iii, 2020-12-05:
Hi - I have resolved this issue.  I contacted support, they suggested I do a factory reset and see the output.  I did and the X axis failed, everything else worked.  They then got me to send videos of the MBL failing which I did, they suggested I undo my belt tension and see if I still had the issue, I undid the belt so is was pretty much close to lose and still it failed.  They suggested the board was faulty and asked me to post it back and they would send the replacement. Problem solved, passed self test and MBL works as expected.

closed 2020-12-05 by jsilva-iii