prusa3d/Prusa-Firmware-Buddy #573: [BUG]<Enter comprehensive title>

Saij, 2020-06-14:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type**
MINI

**Printer firmware version**
4.0.5

**Original or Custom firmware** 
Original

**Optional upgrades**
none

**USB drive or USB/Octoprint**
USB

**Describe the bug**
When switching on the printer with attached USB device I get error #4 (Error mounting USB drive!). After booting I remove the USB drive and reattach it and it works. So there seems to be a problem with the mounting routing on booting.

JohnnyDeer, 2020-06-22:
Hi @Saij , please try this issue on latest FW version 4.1.0-RC1. Should be solved.

JohnnyDeer, 2020-07-13:
@Saij did you tried it on 4.1.0?

Saij, 2020-07-14:
No. Can't get the new firmware to flash...

blsmit5728, 2020-07-14:
I had the same issue going from 4.0.5 -> 4.1.0. Took a few tries but it
finally got it after a fresh format of a drive. (Tried multiple drives etc
before formatting one)

Saij, 2020-07-14:
Already tried with a fresh format and multiple devices. Next try would be network update if that is available.

JohnnyDeer, 2020-07-15:
@Saij you may have faulty boot loader. Please try classic procedure - put FW to format usb stick and unmount USB stick - put stick to printer and hit reset - while Prusa logo is on display stick knob for 5 second - updating should start. If not please try same procedure but with USB extension cable in printer. Another thing you can try is not to put USB stick completely to printer, push it little out. Thank you.

Saij, 2020-07-15:
It worked. Thank you.

closed 2020-07-15 by Saij

seidtgeist, 2020-10-15:
Hey, I’m on `4.2.1-final+2072` and I get this error every startup with the stock USB stick in, no more firmware files on it anywhere (I deleted the one file it came with). Any ideas?

Saij, 2020-10-15:
For me it went away after changing the USB stick. With the one I have right now it works. But updating the system is still a hassle because often it won't detect the new firmware file.

seidtgeist, 2020-10-15:
@Saij Thank you, I’ll try reformatting it or using a different one!

But really, I actually want to print over the network or a USB cable without having to set up OctoPrint. In case anyone here has ideas ;)

Saij, 2020-10-15:
You could use a Raspberry Pi Zero W as a WiFi USB Stick. Just like the FlashAir Cards work.

https://magpi.raspberrypi.org/articles/pi-zero-w-smart-usb-flash-drive

seidtgeist, 2020-10-15:
@Saij _I had no idea_ the Pi (Zero) could act as a USB client. I’ll take it from there, thank you sooo much! 🤯