prusa3d/Prusa-Firmware-Buddy #575: Heating to 170 degrees before print causing confusion

Crawlerin-SA, 2020-06-14:
Is there a way to tell people that heating to 170 degrees Celsius at the beginning of print is preparing for MBL. and not final printing temperature? There are queries appearing every now and then where people - especially new to the printing - stop prints and ask why is it heating to 170 degrees when they set 210 for PLA, thinking it's broken.

This is part of starter G-code in Slicer, thus can be changed. But chances are if guy/gal is new they won't be poking around there much, at least not for first prints. Is there a way to display a comment or a status message on screen (M117) during this process?

This is no burning matter, just a more user (newbie) friendly feature.

JohnnyDeer, 2020-06-18:
Hi @Crawlerin-SA , thank you for suggestion. We know about it and in next FW version this behavior will be improved.

JohnnyDeer, 2020-06-22:
4.1.0-RC1 is out now, with changes touching this FR. Now temperatures on display are blinking with green color, while actual temperature stays on 170°C and desired temperature is shown as seted. I´m closing this issue as solved.

closed 2020-06-22 by JohnnyDeer