prusa3d/Prusa-Firmware-Buddy #586: [BUG] Live Adj (Z Offset) is not saved between reboots for Octoprint "mode"

derenma, 2020-06-17:
**Printer type** - [MINI]

**Printer firmware version**-  4.0.5

**Original or Custom firmware** - Original

**Optional upgrades** - Filament Runout Sensor, Bondtech Extruder, Raspberry Pi w/ OctoPrint

**USB drive or USB/Octoprint** - USB/Octoprint

**Describe the bug** - Live Adj resets to 0.000 after reboot. (Sure, I could set the Z offset manually with G-Code, but unsure if there is custom G-Code that will set Level Adj directly.)

**How to reproduce** - Reboot after setting Live Adj

**Expected behavior** - Live Adj returns to previous setting after a reboot of the printer.

**G-code** - N/A

**Crash dump file** - N/A, Not possible.

derenma, 2020-06-21:
As of version 4.1.0-RC1, Live Adjust is also not saved between Octoprint connect/disconnect/connect cycle. It's slightly different issue, but probably has the same root cause.

EBD232, 2020-07-24:
I had the problem that Live Adjust would always reset to zero when connecting OctoPrint, and in my case it was caused by the OctoPrint plugin EEPROM Marlin Editor. It sends an M501 (Load all saved settings from EEPROM) when connecting, and that resets Live Adjust to 0. I guess it has to do with the new EEPROM implementation in 4.0.5, I didn't fully understand how it is supposed to work. It might still be a bug if M501 resets the Z offset, even though support for Marlin EEPROM is disabled?

jakubmirejovsky, 2020-11-09:
Video for reference: https://www.youtube.com/watch?v=mjnIPwOg624

JohnnyDeer, 2020-12-16:
@derenma this bug is fixed in 4.3.0-RC1 works well when I reproduce it. To be sure, please try it as well and confirm, thank you.

derenma, 2020-12-20:
@JohnnyDeer I have some time today to test, so thanks for the mention!

JohnnyDeer, 2021-02-10:
@derenma 4.3.0 final is already out :) Did you tried it? Would you update this issue please?

derenma, 2021-02-11:
@JohnnyDeer Sweet! I'll get everything tested soon.

JohnnyDeer, 2021-05-06:
I´m closing this issue as solved and stale.

closed 2021-05-06 by JohnnyDeer