prusa3d/Prusa-Firmware-Buddy #603: [ENHANCEMENT]<Bring back sound type from 4.0.5>

Hl2run, 2020-06-20:
Please, before you create a new feature request, please make sure you searched in open and closed issues and couldn't find anything that matches.

Enter what type of printer or upgrade the enhancement applies to.
[MINI]
**Is your suggested improvement related to an existing problem? Please describe.**
  The frequency of the sound when the printer turns on was much much nicer on 4.0.5. The newe 4.1 RC1 has a weird frequency. It sounds as some cheap chinese device. The previous sound was much more pleasant.

x10102, 2020-06-21:
I second this. Just updated and it sounds really like some cheap chinese soldering station or something :D

Dronitoni, 2020-06-27:
same here, the previous sound was much better.
Even if I set it to "silent" mode it makes some noise.

The Mini works in gerneral very silent, that was one argument for me to buy a Mini..
Why adding additional noise to that machine?

4nthonylin, 2020-07-07:
@Chleba was this done in #644 ?

BerenV, 2020-07-12:
I wholeheartedly agree. I can't stand the awful new noises for user interaction. The old ones were so much nicer and felt more clean and professional. Please at least add a mode to switch back to the old ones. It's bad enough that I might even have to downgrade since there isn't anything really useful like Wifi yet 😬 

Thanks!

Dronitoni, 2020-07-12:
my temporary solution is to put a piece of tape on the buzzer. 
This will at least reduce the noise level a little. 
The buzzer is easy to reach on the back of the control unit.

Hl2run, 2020-07-12:
Unfortunately the sound is sounding bad in the final 4.1.0 release :(

jkavalik, 2020-07-12:
@Dronitoni you can switch to silent in settings

Dronitoni, 2020-07-12:
@jkavalik 
I use the 4.1.0 RC1 at the moment.
There is a "silent" setting but it doesn't really switch to silence.

I have not tried the final release yet. Hopefully it is getting better.

jkavalik, 2020-07-12:
@Dronitoni well, the sounds seem the same to me, but the "silent" option works for me at least :)

BerenV, 2020-07-13:
Well my “fix” has been to just downgrade to 4.0.5 😂 I hope the sound gets changed back to the old one though in a future update, or I might have to compile and flash my own modified version 😬

Dronitoni, 2020-07-13:
@BerenV 
4.0.5 could be a solution for the noise, but will bring back the problems with the USB-Stick detection.

BerenV, 2020-07-13:
@Dronitoni I’m using OctoPrint so I haven’t had any issues with the USB drive ;-)

BerenV, 2020-08-18:
Well, 4.2.0 has been pretty good and I appreciate that they added a volume adjustment (although even the lowest setting is still pretty loud). However, the sound is still the same cheap sounding chirp and I have mine set to silent because I couldn't stand it. Sigh...
I wish there was an option to go back to the old subtle but quality sounds for knob rotation and press.

JohnnyDeer, 2020-10-21:
The sounds have already been improved, I close this issue as solved.

closed 2020-10-21 by JohnnyDeer