prusa3d/Prusa-Firmware-Buddy #605: [META] Why don't people close their issues when they are solved?

x10102, 2020-06-21:
Seriously, out of the 80 issues here lots of them are feature requests that have been acknowledged or already added or bugs that have been fixed. 
I think it would be nicer for both the dev team and the users if people closed the issues that are no longer relevant.

3d-gussner, 2020-06-21:
If you find some ask the owner to close these. As development team it always looks "bad" when they close these.
I did it in Prusa-Firmware and some closed it, some had to be closed by Prusa after being stale for months. 

Less open issues makes it way easier to focus on these.

bobc, 2020-06-22:
Will be fixed in Human 2.0 :)

But seriously, this is an unanswerable question that will never be closed....

x10102, 2020-06-24:
> If you find some ask the owner to close these. As development team it always looks "bad" when they close these.
> I did it in Prusa-Firmware and some closed it, some had to be closed by Prusa after being stale for months.
> 
> Less open issues makes it way easier to focus on these.

Yeah, but I don't really wanna be that one annoying guy that reminds everyone to close their issues :D

closed 2020-07-09 by x10102