prusa3d/Prusa-Firmware-Buddy #608: [QUESTION] Why is bed leveling incorporated into Z height on the display?

derenma, 2020-06-22:
Printer type - [MINI]

Printer firmware version- 4.1.0-RC1

Original or Custom firmware - Original/Beta

Optional upgrades - Filament Runout Sensor, Bondtech Extruder, Raspberry Pi w/ OctoPrint

USB drive or USB/Octoprint - USB/Octoprint

Describe the question - Bed leveling adjustments are being shown on screen on 4.1.0. It is important to me to see which layer the printer is working on so I can compare it to the slicer to debug fine prints. 

Expected behavior - I just want to see which layer the printer is working on.

Texy, 2020-06-22:
Yeah I spotted this as well.....

keksa, 2020-07-04:
Just noticed this too. I have literally no use for the real Z value, which is jumping all over the place. I just want to know which layer I'm printing. :)

abrugh, 2020-07-11:
+1 to what everyone else is saying. Although I guess it gives some indication of how not level the bed/gantry are relative to each other? I'm seeing a Z value betwen 5.6 and 6.6 on the same layer, at 0.15mm layer heights. /shrug

Texy, 2020-07-11:
V4.10 has just been released....with this bug still in there unfortunately.

posttenebre, 2020-07-20:
+1 for this issue. I want to know what level I am on to follow the print and spot errors. This issues stops me from being able to do that. Please bring back level height and not absolute z-height.

posttenebre, 2020-08-05:
Has this been fixed in 4.2.0 ?

Gyrxiur, 2020-08-10:
Nope, still there. Do they read this forum?

JohnnyDeer, 2020-08-17:
We have plan to change this feature in some of future releases. Thanks for patience.

matejus, 2020-10-24:
Hi all, for me seems to be good solution to show printed layer (number or height) next to percents of print, there is free space 
e.g. 9% (L 0.45mm)

mivanov, 2021-08-30:
+1 for this. Still seeing it in 4.3.2. Seeing the real Z height has a lot of value but probably not as much as the G code one or the level number. Love my Mini+ btw and thank you for the hard work!

Eldenroot, 2022-11-26:
@Prusa-Support - this is done not? In 4.4.0 you can set both values into footer

Prusa-Support, 2022-12-10:
Yes, thanks.
This issue is directly mentioned in the [FW 4.4.0 release notes](https://github.com/prusa3d/Prusa-Firmware-Buddy/releases/tag/v4.4.0) too.
Closing now.

Michele Moramarco
Prusa Research

closed 2022-12-10 by Prusa-Support