prusa3d/Prusa-Firmware-Buddy #609: Config Menu for Sensorless Homing Sensitivity

ChengQian1007, 2020-06-22:
Could you please create a menu for config the sensorless homing sensitivity?

x10102, 2020-06-26:
Why do you need this? If the sensorless homing doesn't work correctly you probably got a hardware problem.

ChengQian1007, 2020-06-27:
Reassemble the Axises, belt tension, bearing, grease, not parallel X Axis and so on, all these could lead to increase the fiction and have influence on the Sensorless Homing Sensitivity value and thus result the failure to homing. Cannot home can not print easily. It is also much more easier to tune the sensitivity value a litte bit as to find out which one component of many leads to the problem.
I have also read in the forum that some printers have this problem directly after arrival.

closed 2024-07-04 by github-actions[bot]