prusa3d/Prusa-Firmware-Buddy #626: [Bug] Reopening: Remote printing w/ Octoprint - filament change not working

ifekali, 2020-06-25:
This issue (#362) was closed prematurely, I am reopening it here as follows:

> Further to issue #360 during remote printing when the printer reaches the M600 command in GCODE
> extruder pulls out the filament and waits for user input (a.k.a. continue when the new filament is loaded).
> 
> So, while doing that, when the new filament is loaded and I press continue, the printer comes back to the remote printing > screen without loading the filament. Funny thing is it doesn't come back to print either.
>
> Prusa mini FW version 4.0.5, Prusa slicer 2.2.0

This is still an issue with 4.1.0-RC1 firmware, confirmed by at least two other users.

**Original Prusa Mini** 
**4.1.0-RC1 firmware**

deltwalrus, 2020-06-26:
Confirm this is still an issue with 4.1.0-RC1

Chleba, 2020-07-09:
This bug have been fixed in v.4.1.0 final version.

twobit0, 2022-11-07:
This is still an issue in 4.3.4!!! can we reopen this?

Prusa-Support, 2022-11-25:
I'm sorry for the confusion.

Because it is confirmed that the issue showed up again, I will actually close this one as a duplicate and reopen https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/362.

Michele Moramarco
Prusa Research

closed 2022-11-25 by Prusa-Support