prusa3d/Prusa-Firmware-Buddy #632: [BUG] Flash drive sometimes needs to be reconnected

x10102, 2020-06-26:
**Printer type** - MINI

**Printer firmware version**-  4.1.0 RC1

**Original or Custom firmware** Original

**Describe the bug**
Since updating to 4.1.0 RC1  Sometimes the printer shows a "no USB" message after turning it on even though USB is plugged in. Removing and re-inserting the USB fixes this.

**How to reproduce**
  1. Insert the USB drive
  2. Shut down the printer
  3. Turn the printer back on

**Expected behavior**
  The USB should always show up when inserted

JohnnyDeer, 2020-06-29:
Hi @x10102 , thank you for reporting. Are you using original usb stick?

x10102, 2020-07-09:
@JohnnyDeer Yes I am using the original stick.

JohnnyDeer, 2020-07-13:
@x10102 unfortunately i can't reproduce it. Did it happens to you also on 4.1.0 final?

x10102, 2020-07-14:
It doesn't, nice.

closed 2020-07-14 by x10102