prusa3d/Prusa-Firmware-Buddy #635: Nozzle heater disabled after filament change mid-print

Texy, 2020-06-27:
[MINI]
It seems the 15(?) minute nozzle heat has been switched off whilst waiting for new filament to be inserted.
So whenever cgode requests a filament change, the old is ejected and the nozzle heater is disabled immediately.
When you press the button to insert the new filament the nozzle heater is re-enabled, but time is wasted waiting for reheat.
Previously the nozzle was kept to temperature until the auto-timeout disabled it after 15(?) minutes.

JohnnyDeer, 2021-05-19:
@Texy this is not happening in latest FW version. Can you please confirm it and close this issue?

Texy, 2021-05-19:
No problem. I haven't noticed it for a long time, so guess it was fixed.

closed 2021-05-19 by Texy