prusa3d/Prusa-Firmware-Buddy #637: [ENHANCEMENT]First Layer Calibration Finish Position

tjhadley, 2020-06-28:
Related to #50 and #242 . Prusa Mini. Running firmware 4.0.5.

When a typical print finishes, the bed returns to (x,y) 170,170.

When first layer calibration finishes, it returns to (170,0) This puts the bed at the rear-most 
position that makes clearing the filament off more difficult. I end up moving the axis forward and raising the z-height in order to clear off the calibration filament. 

Can the default after-calibration home position be set to (170,170) like it does for normal prints?

evoncken, 2020-11-03:
As a user, I want to take off the steel print sheet after calibration or printing so I can clean it. The current default end position makes this impossible. Would it be possible to change the default end position?

ftrofin, 2020-11-18:
OMG, I am so happy I'm not the only one thinking that this is a must! I just searched the open issue before writing the exact thing and I'm glad that somebody else thinks this is nuts!
Prusa Devs, please, run the calibration step 10-15 times in a row (like I did two days ago) and you'll see very quickly what we mean :-) This, combined with a better calibration pattern (see #832 ) would make my life soo much easier with my Prusa Mini!

I suggest moving the head and the plate at the end of calibration to the same positions where they go when you cancel a print and disconnect from OctoPrint (the Z moves up significantly higher and the plate comes forward) which makes lifting the plate and cleaning it much easier and without danger of getting burned.

ftrofin, 2020-11-18:
PR created for this, please review and merge.

closed 2020-11-20 by RadekVana

ftrofin, 2020-11-21:
Thank you @RadekVana !

PR #1063 (2020-11-20): Park the head after calibration.
Move the head higher on Z axis after calibration is done so the plate can be cleaned easily. Fixes #637.
commit 0408a178 (2020-11-20): Park the head after calibration. (#1063)

Move the head higher on Z axis after calibration is done so the plate can be cleaned easily. Fixes #637.