prusa3d/Prusa-Firmware-Buddy #648: [ENHANCEMENT] Please bring back relative remaining time

noxhirsch, 2020-07-02:
**Printer type** - [MINI]

**Is your suggested improvement related to an existing problem? Please describe.**
  With the new release candidate the expected end time is calculated from relative remaining time. While that is really neat when selecting a file it gets pretty annoying while the print is running. I always have to check the clock to get a feeling of how many minutes remaining - or I have to "calculate" the remaining minutes in my head using the printing time and percentage.

**Describe the expected functionality**
  For me it would be perfect to have the absolute end time only in the file preview screen before printing. In the printing screen relative remaining time would be enough.
Alternatively the printing screen could show both types of time e.g. above each other (but not switching because then you check the display at the wrong moment half the time) or it could be an option.

caro-coder, 2020-07-02:
I'll second that! It drives me nuts! 🤯

nicolasvac, 2020-07-08:
I agree with that. Put both infos, like 2h 58 minutes remaining (23:58)

mlgold, 2020-07-09:
I'm would prefer to have either the estimated completion clock time and minutes and seconds remaining, or the estimated completion time and the current time. This should be on both the LCD and web page.

colingg, 2020-07-13:
Give us the option as a setting please:

1. Show ETA time ( eg: 14:05 )
2. Show relative time ( eg: 27 minutes remaining )
3. Show both: ( eg 27 minutes remaining [14:05] )
4. Switch between 2 times **1 AND 2**

adynis, 2020-07-24:
I vote for showing both (or at least (additional to ETA time) to add current time in the top, like any mobile phone)

Gyrxiur, 2020-08-10:
Yeah, show both, screen is large enough.

Ondrashek, 2021-01-04:
I want relative time too. :-)

anymouse404, 2021-02-14:
This problem affects me as well. 
Currently there is no decent way to know what time the MINI is set to, and if it gets confused by DST (and it does) then the estimate is off by an hour, which is very frustrating.

chillygithub, 2021-03-30:
Another vote for relative time.
I have resorted to blocking NTP from the Mini at my router so that the relative time is shown.

Eldenroot, 2022-11-23:
Showing both! So number 2)!

Eldenroot, 2022-11-23:
Also the worst thing - if you connected to the internet (4.4.0_RC1) - the time is switched to relative... annoying!

Prusa-Support, 2022-12-01:
Thank you for sharing your ideas.
As the latest firmware 4.4.0 will possibly "democratize" the connection capabilities, and leverage related features, we may see an increasing interest in the topic.

We are taking notes and discussing them with the developers.

Michele Moramarco
Prusa Research

DrBomb, 2023-04-14:
I hope it can be brought back.

dimitarsp, 2023-05-28:
I recently upgraded the firmware, and yes, like everyone else, I desparately miss seeing the remaining print time on the prusa mini display. Please bring this back!

Prusa-Support, 2023-09-29:
I may seem to understand that the behavior is different with the new FW 5.1.0-alpha1.

The relative remaining time is back and displayed even if the printer is connected to the network.
The estimated end time is available through Prusa Link/Connect interface.
I can't tell if this will be definitive though so the issue can stay open for the time being.

Michele Moramarco
Prusa Research

Prusa-Support, 2023-09-29:
(you may receive email notifications but the last two messages will be deleted)

That was totally my bad.
Later I noticed the wi-fi signal was not stable on the router.
Now it is stable and I can see the print end time and not the remaining time.

Michele Moramarco
Prusa Research

espr14, 2023-12-18:
### Like in [Snake FW](https://github.com/Snake-FW/P32-FW/releases)?

[image]

Prusa-Support, 2024-02-22:
I seem to understand this request has been implemented and the commit is probably this one as the time information will appear in a cyclical alternation https://github.com/prusa3d/Prusa-Firmware-Buddy/commit/cc4936cee1c0ba7a7c152a3ebb6ca5c5f8f77cbb.

However, it won't be available until the next firmware public release, which will probably occur in several weeks or even a few months as there are big projects in progress. We don't have an estimated time but please stay tuned.

Michele Moramarco
Prusa Research

Prusa-Support, 2024-05-23:
The commit that addresses this issue was merged and included already in FW 5.2.0 (MK3.5 only) and spread on MK4/MK3.9/XL printers with the more recent FW 6.0.0.

The issue will be closed as solved.

Michele Moramarco
Prusa Research

closed 2024-05-23 by Prusa-Support

anymouse404, 2024-05-23:
> [...]included already in FW 5.2.0 (MK3.5 only) and spread on MK4/MK3.9/XL printers[...]

> The issue will be closed as solved.

Does that mean that it's acutally a WONTFIX for the MINI that *this* issue is about?

Prusa-Support, 2024-06-15:
That was my mistake in the listing. The same firmware base code is shared with Prusa MINI so the change should apply to all "Buddy-based" products. I've corrected my previous comment.

Michele Moramarco
Prusa Research